bastibl/gr-ieee802-11 #99: wifi_loopback.py is not going through ieee802_11_decode_mac (171/177 symbols)

lo-bert, 2018-02-06:
Hi Bastian, thanks for your effort in reading this Issue. 
After the preliminary steps, building **_wifi_phy_hier.grc_** with some debug options active, and running **_wifi_loopback.grc_**, I notice that no messages reach the **ieee802_11_parse_mac** block. 
This because the receiver block **ieee802_11_decode_mac** can't decode any MAC layer packet. It always receives only 171 symbols over 177 (for PDU = 500 / 528 with frame, at the sender). This repeats over and over again.

additional **ieee802_11_decode_mac** debug output is the line (at each new packet beginning parsed): 
`Decode MAC: frame start -- len 528  symbols 177  encoding 0`

May you help me in fixing the issue? Thanks,
Lorenzo

[image]

lo-bert, 2018-02-06:
Additional Info:

 

- I can correctly receive 802.11 MAC packets over the Radio interface with **_wifi_rx.grc_**

- radio-config-info--version
3.7.9

bastibl, 2018-02-07:
Maybe you modified the flow graph? It's not in the picture, but since you don't have Wireshark block installed, I assume that you also deleted the Packet Pad block. Please make sure the unmodified flow graph works.

lo-bert, 2018-02-08:
Hi Bastian, thanks for your response.
A clean Installation of the master branches fixed it. Sorry for the useless thread. 
I'll close it.

closed 2018-02-08 by lo-bert