bastibl/gr-ieee802-11 #100: in wifi_transceiver getting RuntimeError: invalid msg port in connect() or disconnect() issue

harshitk26, 2018-02-14:
Hi
I am running wifi_transceiver with USRP e312 and in runtime I am getting bellow issue:

Traceback (most recent call last):
  File "/home/harshit/gnuradio/gr_pkg/gr-ieee802-11/examples/wifi_transceiver.py", line 352, in <module>
    main()
  File "/home/harshit/gnuradio/gr_pkg/gr-ieee802-11/examples/wifi_transceiver.py", line 340, in main
    tb = top_block_cls()
  File "/home/harshit/gnuradio/gr_pkg/gr-ieee802-11/examples/wifi_transceiver.py", line 172, in __init__
    sensitivity=0.56,
  File "/home/harshit/.grc_gnuradio/wifi_phy_hier.py", line 83, in __init__
    self.msg_connect((self.ieee802_11_decode_mac_0, 'out'), (self, 'mac_out'))
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 59, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 131, in msg_connect
    self.primitive_msg_connect(*args)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3570, in primitive_msg_connect
    return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()

any solution, I am running wifi_transceiver without any changes 

Thanks

bastibl, 2018-02-14:
... and you created the hierarchical block for the PHY as mentioned in the readme?

harshitk26, 2018-02-15:
Hello Bastibl,
yes I did, after that ... .grc_gnuradio/wifi_phy_hier.py is created

bastibl, 2018-02-15:
...strange. Any chances you use a very old GNU Radio version? Maybe you can open the flow graph (and the hierarchical block) in GRC. Don't know the E312, but maybe you can forward X or try a similar setup on the desktop.

JgrisaleC, 2018-05-22:
Hello, 

I've had the same error when running wifi loopback.grc, my GNU Radio Version is 3.7.9, I installed it using Pybombs. I also created the hierarchical block for the PHY.

Executing: /usr/bin/python -u /home/juanpablo/gr-ieee802-11/examples/wifi_loopback.py

MAPPER: encoding: 0
set_min_output_buffer on block 7 to 397056
Using Volk machine: avx_64_mmx_orc
set_min_output_buffer on block 3 to 397056
set_min_output_buffer on block 5 to 397056
set_min_output_buffer on block 6 to 397056
set_min_output_buffer on block 8 to 397056
Could not find port: out in:
system

Traceback (most recent call last):
  File "/home/juanpablo/gr-ieee802-11/examples/wifi_loopback.py", line 330, in <module>
    main()
  File "/home/juanpablo/gr-ieee802-11/examples/wifi_loopback.py", line 318, in main
    tb = top_block_cls()
  File "/home/juanpablo/gr-ieee802-11/examples/wifi_loopback.py", line 140, in __init__
    sensitivity=0.56,
  File "/home/juanpablo/.grc_gnuradio/wifi_phy_hier.py", line 81, in __init__
    self.msg_connect((self.ieee802_11_decode_mac_0, 'out'), (self, 'mac_out'))    
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 59, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/hier_block2.py", line 131, in msg_connect
    self.primitive_msg_connect(*args)
  File "/usr/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 3584, in primitive_msg_connect
    return _runtime_swig.hier_block2_sptr_primitive_msg_connect(self, *args)
RuntimeError: invalid msg port in connect() or disconnect()

bastibl, 2018-05-22:
It looks like you installed GNU Radio through PyBombs, but the module yourself. I guess something went wrong there. Try installing also the module through PyBombs. I guess you have multiple GNU Radio versions on your system.