bastibl/gr-ieee802-11 #101: Documenting the meaning of local oscillator (LO) offset and its value

frestuc, 2018-02-20:
It is not clear why we use the LO offset values of 6e6 and 11e6 to tune the LO offset, and what is the purpose of this parameter. Could we investigate more on the topic and then add some comments on the examples? Thanks.

bastibl, 2018-02-21:
I'm not sure what's not clear. LO is the abbreviation for the local oscillator. Using LO offset tuning, you can move the LO out of the signal of interest. The USRP is a [direct conversion receiver](https://en.wikipedia.org/wiki/Direct-conversion_receiver) which suffers from a [DC offset](https://en.wikipedia.org/wiki/Direct-conversion_receiver#Disadvantages).  See also the [application notes](https://files.ettus.com/manual/page_general.html) from Ettus. 
With offset tuning, the receiver tunes to a frequency out of the band of interest and shifts frequency in digital domain. For example, instead of tuning to the center of a 10MHz channel, you use a 6MHz offset to move the LO 1MHz out of the signal of interest. Same for a 20MHz channel and an LO offset of 11MHz.

frestuc, 2018-02-21:
Ok, thanks, now it's clear. Hope this will clarify the issue for other people.

closed 2018-02-21 by frestuc