bastibl/gr-ieee802-11 #103: Overflow

dverhaert, 2018-02-28:
Hey Bastian,

I'm using your GNURadio blocks with just one USRP N210 as a transceiver  Eventually I'd like to do some tests for 802.11.ah, but for now I can't even get the basic wifi_transceiver to work correctly. Initially, I had the problem where GNURadio couldn't allocate any memory for TUNTAP, but I fixed this by just disabling the block. However now, even with the basic 5MHz I keep getting an overflow in messages "DDD....". What I have done to try to fix the problem already is:

- Ran volk_profile
- Installed CorrectIQ block to correct for DC offset
- Doubled maximum shared memory

I'm running a laptop with a i7-7700 HQ processor, giving my Linux VM 10GB of ram and 6 CPUs, so I would think that is good enough. Would you have any idea why I keep getting this overflow? Note that if I decrease my sampling rate to about 1MHz, I don't see any overflow errors. 

My workspace:
[image]

Note that I have an antenna in both RF1 and RF2 pointing in the different direction. Does GNURadio automatically use one for transmitting and the other one for receiving? This might be related to the problem.

My result:
[image]

Any help would be appreciated!

bastibl, 2018-02-28:
Hmm, looks like you tried the most complex setup first... How about starting with a simpler one. Does, for example, the loopback flow graph work? Do TX and RX flow graphs work (the separate ones)? Did you try installing the next branch of GNU Radio and this module (there are several scheduler patches in GNU Radio next).  Try an LO offset. Try a faster PC (or a different network card).

dverhaert, 2018-02-28:
I checked for the simpler ones. Loopback works without a hinch. Wifi_tx has some underflow messages (maybe 1-3 per second), but the console isn't overflooded. As soon as I start running wifi_rx however my console gets flooded by overflows and I get an output like this:

```
DLONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 320
DDLONG: frame start at 320
DDDDDLONG: frame start at 320
DLONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 320D
DLONG: frame start at 117
DLONG: frame start at 320
DDLONG: frame start at 320
DDLONG: frame start at 320
LONG: frame start at 320
DDDLONG: frame start at 320
DLONG: frame start at 320
DLONG: frame start at 320
DDDLONG: frame start at 320
DLONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 152
LONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 320
DLONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 320
LONG: frame start at 178
DDDDLONG: frame start at 320
```
It gets better when I add a correctIQ block, but the problem still perseveres. I also tried setting an LO offset (no effect), and don't really have access to a faster PC at the moment (just bought this new laptop and thought it'd be pretty fast). Could it be my VM? I've tried putting all the settings as high as possible, e.g. just enabled 3D acceleration as well but still no luck.

I did not try installing the next branch yet, would your next branch work work with the PyBOMBS version of GNUradio? When I simply installed GNUradio by apt-get before I couldn't get my USRP working since there always was some version mismatch. 

For now the only thing that removes the overflow errors is sampling at 1 MHz, but this leads to incorrect results (I'd guess because the sampling rate is not high enough).

bastibl, 2018-02-28:
...so you are running inside a VM. Some people forget to enable HW acceleration for VMs. Some don't compile the required kernel drivers for native drivers. Some use bogus setting in their VM. Some just allocate 1 or two CPU cores and basically no memory (3D acceleration is usually no problem). And some appliances just don't manage to pipe 20 Million complex floats into the VM and decode in real-time. That's why the installation instructions on https://www.wime-project.net/installation/ say "Run the transceiver on a native installation (no VM)." I don't say that it's not possible to run it in a VM, but I wouldn't recommend it if you are new to SDR.

dverhaert, 2018-03-01:
I am running inside a VM. Enabled hardware virtualization with a suitable paravirtualization interface, checked my settings: allocated 7 cores and 14GB of RAM to my Virtualbox. Can you think of anything else I could try to speed up the system? Or maybe another reason besides the sampling rate being too high (even when I set it to be 1 MHz) why my results keep being randomly spread dots?

I'd like to compare the performance results of my Windows machine and inside my VM with a benchmark, but I can't seem to find any tool running on both. Any idea how I can check if it's really my Virtualbox performance? What numbers should I expect with a stress-ng/sysbench test?

Otherwise I fear you are right and I must go through the process of installing everything once more, this time using a desktop. I just think the solution could be so simple and save me the hours of reinstallation...

dverhaert, 2018-03-04:
I tried to debug the ethernet driver on my VM for a while, but to no avail. Consequently I managed to acquire a desktop and installed Ubuntu natively, and the overflow error is gone, thanks! I do have another problem however, will post in another issue.

closed 2018-03-04 by dverhaert