bastibl/gr-ieee802-11 #104: Incorrect output

dverhaert, 2018-03-06:
Hi Bastian,

I've acquired a second USRP N210, and have tried testing with these devices on a sufficiently fast desktop with native Linux install and a network card/cable supporting 1Gbit/s. I've tried the following setups:

- Device 1 running wifi_transceiver
- Device 2 running wifi_transceiver
- Device 1 running wifi_tx and wifi_rx
- Device 2 running wifi_tx and wifi_rx
- Device 1 running wifi_tx, device 2 running wifi_rx
- Device 2 running wifi_tx, device 1 running wifi_rx

Unfortunately, none of these showed any correct results. I checked the traffic over the cable for debugging with **bmon**. First of all, when I run wifi_transceiver, the program shortly transmits and receives but then resorts to only receiving. When I run wifi_rx and wifi_tx at the same time, I noticed monitoring the network traffic that as soon as I run the second program, the first one stops (e.g. I have wifi_rx running and run wifi_tx, after which I notice on bmon and by the lights on the device that it stopped receiving and so is only transmitting). Finally, when I let one device transmit and the other receive, I still don't get any results. 

I've tried so many things the past couple of days, would you have any idea what's wrong? If one of the devices is not working, I would say the other would at least work as a transceiver (built my own which for some reason is able to both send and receive at the same time), but unfortunately this too doesn't work as shown in [1]. 

Since I figured this might be a hardware issue, I tried going back to a basic example running on both devices as shown in [2] and [3]. Suprisingly, I get very different results with exactly the same measurement. Do you think any of these devices is broken? I'm really out of clues, any help would be immensely appreciated. Thanks in advance!

[image]
[1]

[image]
[2]
[image]
[3]

bastibl, 2018-03-06:
If you want to send and receive, you should use the transceiver flow graph. However, I'm not sure what you expected from it. If you didn't send something, it's obvious that the flow graph just received.
Anyhow, before setting up the transceiver, you should first get RX and TX run separately.

The plots are obviously different. Maybe one uses a different daughtboard, has an opened case, or is placed somewhere strange.
I don't understand why you don't start simple and just use your WiFi card (in 5GHz) to send and receive frames from the SDR (in separate experiments). This will tell you if zero, one, or both devices work.

dverhaert, 2018-03-07:
In the end, I figured out the power adapter of my N210 was delivering a too low voltage, messing up with my measurements. After debugging with my wifi card I got the  other one working by changing the gain from normalized to absolute!

I get outputs at 2.4G now, but it's still kind of noisy. Would you know of any ways to improve the system to reduce noise? Also, do you know if anything special is required to make it work for sub 1 GHz frequencies? Currently can't get it to work for those. 

Finally, is it doable to add a new modulation technique (256-QAM)? If so, what could be a good starting point?

Thanks again, you've been really helpful!

bastibl, 2018-03-07:
You could experiment with different LO offsets. Also, since you mentioned that switching to absolute gain did the trick, suggests that you should experiment with different gain settings. You can either switch to normalized and set a value between 0 and 1 (easiest way) or set an absolute value (however, the range depends on the daughterboard that you plugged in). 

To add other modulations you have to adapt quite some parts of the code but you might want to start looking into the receiver (frame equalizer) or the standard to understand how the MCS is signaled.

dverhaert, 2018-03-16:
Experimenting with different LO offsets, gains, and some other small optimizations did it for me.

In the name of everyone who you've helped with issues these past years: thanks for always quickly replying with helpful feedback!

closed 2018-03-16 by dverhaert