bastibl/gr-ieee802-11 #106: Help：Unable to send frames greater than 1500

wshe1, 2018-04-10:
Hallo, Bastian.
I modified the program to expect to use the Socket PUD to send a frame length of 5000. However, a maximum of 1500 frames can be sent. I changed the max payload from 1500 to 5500, but it can only send 1500.The following is my modified program, what should I do?
![Uploading Screenshot from 20
[image]
18-04-09 23-36-49.png…]()

bastibl, 2018-04-10:
Looks good.
What do you mean with it doesn't work? Is there a warning or an error? Are the block buffers too small? Where in the TX chain does it break? Are you shure it breaks in TX (an not in RX)?

wshe1, 2018-04-10:
Hallo, Bastian.
The problem just solved, under the solution method。But I encountered your new problem and I need your answer.
[image]
[image]
[image]

Nattaro13, 2018-04-13:
Hallo Bastian,

We have a similar problem as @wshe1 even after we have pulled the fixes and made the min output buffer sizes much larger in the corresponding python scripts as well. 

```
sched: <block chunks_to_symbols (9)> is requesting more input data
  than we can provide.
  ninput_items_required = 80496
  max_possible_items_available = 65535
  If this is a filter, consider reducing the number of taps.
```

Could we get your help on this as well?

Thank you!

bastibl, 2018-04-17:
You have to increase the buffer size. As I said:

> The buffer sizes can be changed in GRC (Advanced -> Min Output Buffer Size)

closed 2018-04-17 by bastibl