bastibl/gr-ieee802-11 #109: Using wireshark

jabirshabbir, 2018-04-17:
Hi,
Is it indeed necessary to use the next branch of your module for wireshark and file dump.If possible can you suggest any alternative to this as i just need information like SNR estimate and  packet capture

bastibl, 2018-04-17:
Please see the readme:

> Like GNU Radio, this module uses master and next branches for development, which are supposed to be used with the corresponding GNU Radio branches. I recommend staying up-to-date by using the next branch.

So you can use the `master` branch of this module (and gr-foo) with the `master` branch of GNU Radio.

closed 2018-04-17 by bastibl

jabirshabbir, 2018-04-17:
Thanks.So Can I get the wireshark components working by installing master branch of your module with master branch of gnuradio?>I may be wrong but according to my understanding the gnuradio installed from sudo apt get and master is the same? Moreever the master module of your project with wiresharsk components enabled in RX.grc didnt run with gnuradio installed with sudo apt-get

jabirshabbir, 2018-04-17:
Alright thanks a lot