bastibl/gr-ieee802-11 #113: A question about wifi_rx.grc example

totoshka804, 2018-04-27:
Hello Bastian,
sorry about my poor English. I have read your paper “An IEEE 802.11a/g/p OFDM Receiver for GNU Radio” and now I’m trying to implement some sort of wifi scanner based on your project using an example wifi_rx.grc. In this flowgraph “Wifi parse MAC” block finds control and data frames (between clients and the access point), but it seems that regardless of the parameters used, this block can’t find any management frames. What could be the cause of this problem in my case?
Thank you for helping, best regards.

bastibl, 2018-04-27:
Hi, the WiFi Parse block is more for debugging purposes then for performance evaluation. It doesn't implement a full WiFi parser. I'd recommend to use the Wireshark Connector and create a PCAP file that you can inspect with Wireshark. There you should see all frames  properly parsed.

bastibl, 2018-04-27:
This is really strange, since from the receivers perspective, management frames are in no way different. Make sure that the frames you are looking for are actually using IEEE 802.11a/g/p, i.e., make sure that they are not using 11b, for example.
I guess the easiest way is to use an 11a AP in the 5GHz band. There shouldn't have any issues with PHY modes.

closed 2018-05-02 by bastibl