bastibl/gr-ieee802-11 #115: Example block diagram freezes

ggamble22, 2018-05-16:
When running the receive block diagrams, the system stops receiving data after a a variable time period ranging from maybe 5 to 30 seconds. I am using a USRP X310 with a UBX-160 daughter board. I ran all of the uhd_cal* programs and volk profile. I am running Ubuntu 18.04 but also saw it on 17.10. I am currently using gnuradio 3.7.11 but I got the same results by compiling 3.7.12. Let me know if there is any other info that would be helpful.

bastibl, 2018-05-16:
There are multiple examples, which one are you referring to?

What do you mean with stops receiving? Does the flow graph terminate? If the flow graph is still running, find out where things go wrong.  Disable components at a later stage and assert, for example, that the SDR still streams samples. In the beginning you could start with the SDR and some GUI stuff only. Then add more blocks back.
Furthermore, most of my blocks have log and debug options. They could be helpful to understand which block is causing problems.

ggamble22, 2018-05-16:
The flow diagram is still running but not receiving any data. Also the light on the USRP itself that indicates that an antenna is active goes off. I also have run other block diagrams with it without issues which is why I'm asking here. I will try the block disabling to try and track down the culprit. 

Where are the logs placed so I can review them. There were no error messages with I looked in the console output and with debug turned on for all of the blocks.

ggamble22, 2018-05-16:
It appears to be caused by the Wifi Sync Long block. Starting from the end of the wifi_rx.grc block diagram, I kept having the issue until I got that block. The last output from the block is:
`produced : [some #] consumed: [some #]`
The numbers aren't consistent but it is just like all the other output from the block.

bastibl, 2018-05-16:
hmm, no idea. I never had this problem. You could try different LO offsets or recompile the module, but I doubt that this will help. Also make sure there are no other blocks in the flow graph (i.e., that you didn't add any custom blocks).
If that doesn't help, make sure that it's really the sync long block, i.e., that it stalls even if there is only a null sink is attached. In that case, you might have to add more debug output to understand what's going wrong.

ggamble22, 2018-05-22:
When I was testing, I had was the null sink block attached to the sync long block. For this test, it was just the wifi_rx.grc example that was run unmodified, except for the disabled blocks. Where are the log files stored again so I can look into it and maybe attach it to see if you see something I am missing?

ggamble22, 2018-05-22:
Also, I am currently using a version of the module that I compiled.

bastibl, 2018-05-23:
By default, information is logged to the console. It can be configured like this:
https://www.gnuradio.org/doc/doxygen/page_logger.html

You could try to build GNU Radio and the module with pybombs:
https://github.com/gnuradio/pybombs
This should rule out any compile and installation issues.

ggamble22, 2018-05-24:
Okay, I thought there was a separate log file. I have been seeing the logs in the console and there has been nothing special.  I have seen that lowering the gain seems to increase the amount of time it runs for. Even setting it to 0 causes it to run for 1 to 2 minutes but it still freezes.

Also, I did compile it with pybombs.

bastibl, 2018-05-24:
That's really strange... I'm afraid I'm running out of ideas. I never had the chance to test it with an X310. Could you get your hands on a B210 or N210?

ggamble22, 2018-05-24:
Unfortunately I do not have one available but I have been successful with setting the clock rate from default to 120 MHz. So far I've had it running for 20 mins without a problem so I will close this issue for now. Thanks for all the help.

closed 2018-05-24 by ggamble22