bastibl/gr-ieee802-11 #116: Static OFDM Allocator

davidzwa, 2018-05-22:
I'm involved in creating a 802.11ah model in GNUradio and I was directed to your model.
It is very nice indeed!

Since the required changes are substantial (carrier frequency, OFDM carriers, bandwidth, MCS, PHY, MAC. list goes on) I am trying to focus on the most important things first.
Is it true that the OFDM carriers are static in your block example?
Is it true that the receiver side of the PHY block (autocorr, sync short, sync long, frame equalizer) are dependent on the OFDM being the same (hardcoded)?

I am planning to rewrite the OFDM allocator block from GNUradio, since it doesn't support any of my OFDM carrier plans. Can you give me some hints on what functions I should be careful with to change to, for example, travelling pilots or different pilots altogether?

I understand this is a lot to ask, so you can keep it to the point to not waste your time

bastibl, 2018-05-22:
You're right. This is a very application-specific implementation and far from a general-purpose OFDM implementation. It sounds like the only similarity between the PHY is that they are part of the same standard :-)

I'd recommend to use the OFDM implementation of GNU Radio as a base. It's designed to be a generic implementation that can easily be adopted.

davidzwa, 2018-05-25:
Ok, I'll try and write a configurable OFDM allocator and fork from your repo so you can merge it if you're interested

closed 2018-05-25 by davidzwa