bastibl/gr-ieee802-11 #120: Can't build module on ARM

creffett, 2018-06-20:
Hi,

I'm trying to build this on a Raspberry Pi, but make fails with the following:
`[  7%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/moving_average_cc_impl.cc.o
c++: error: unrecognized command line option '-msse2'
`

Presumably, the issue is that ARM doesn't have -msse2. I commented out the msse2 line from CMakeLists.txt and I get a different (also x86-specific) error:
`In file included from /root/sdr/gr-ieee802-11/lib/decode_mac.cc:20:0:
/root/sdr/gr-ieee802-11/lib/viterbi_decoder.h:20:10: fatal error: xmmintrin.h: No such file or directory
 #include <xmmintrin.h>
          ^~~~~~~~~~~~~`

bastibl, 2018-06-21:
Unfortunately, I don't have an ARM setup, so I cannot test it myself. But you shouldn't comment out anything. 
The plan is that cmake first [checks if sse is available](https://github.com/bastibl/gr-ieee802-11/blob/next/lib/CMakeLists.txt#L93) and based on the result either uses the [accelerated or the generic implementation](https://github.com/bastibl/gr-ieee802-11/blob/next/lib/CMakeLists.txt#L126-L135). Maybe you can debug the cmake variables and make sure that the test worked as expected, i.e., that SSE2 is not enabled.

Also, your error is very strange since `gr-ieee802-11/lib/viterbi_decoder.h:20` [does not include anything](https://github.com/bastibl/gr-ieee802-11/blob/next/lib/viterbi_decoder/viterbi_decoder.h#L20). Did you  change the sources?

creffett, 2018-06-21:
That doesn't sound like what I have. A coworker handed me this to build for a project, let me make sure that he didn't do something weird like check out an ancient branch.

creffett, 2018-06-21:
Aha! It's because I'm on the master branch. The next branch builds fine.

bastibl, 2018-06-21:
Great. Then I'm closing the issue, since the master branch will be obsolete soon (when GNU Radio merges its next branch into master, which is currently prepared).

closed 2018-06-21 by bastibl

dwalexander, 2019-01-08:
Hi there,

I am currently trying to install gr-ieee802-11 on an Ettus E312 (along with gr-foo). The gr-foo was able to be installed successfully on the master branch. However, with gr-ieee802-11, after running `cmake ..` successfully, when I tried running `make` as per the instructions I have got the same issue with the `-msse2` flag being unrecognised. I am running the master branch of gnuradio (on the latest sd card image for the E312) and attempting to install the master branch of gr-ieee802-11. I have also tried the next branch to no avail. Any thoughts as to any solutions?

For reference here is the trace of the failed compilation
`[  9%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/moving_average_cc_impl.cc.o
c++: error: unrecognized command line option '-msse2'
lib/CMakeFiles/gnuradio-ieee802_11.dir/build.make:70: recipe for target 'lib/CMakeFiles/gnuradio-ieee802_11.dir/moving_average_cc_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/moving_average_cc_impl.cc.o] Error 1
CMakeFiles/Makefile2:153: recipe for target 'lib/CMakeFiles/gnuradio-ieee802_11.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/all] Error 2
Makefile:123: recipe for target 'all' failed
make: *** [all] Error 2`

rcetin, 2019-03-28:
Hi Bastian,
I am trying to compile gr-ieee(master branch) for E310 (ARM). I guess there is no next branch in gnuradio so i tried to compile master branch for ARM. My problem is same with this issue so i commented here. 
Compilation gives that error.

`rcetin@rcetin-W540:~/e300/src/gr-ieee802-11/build [±|master ✓|] $ make -j8
Scanning dependencies of target ieee802_11_generated_includes
Scanning dependencies of target ieee802_11_swig_swig_doc
Scanning dependencies of target pygen_python_7d8a9
[  8%] [  8%] [  8%] [  8%] Built target ieee802_11_swig_swig_doc
Generating moving_average_ff.h, moving_average_cc.h
Generating __init__.pyc
Generating __init__.pyo
[  8%] [  8%] Built target ieee802_11_generated_includes
Built target pygen_python_7d8a9
Scanning dependencies of target ieee802_11_generated_sources
Scanning dependencies of target _ieee802_11_swig_swig_tag
[ 14%] [ 14%] Building CXX object swig/CMakeFiles/_ieee802_11_swig_swig_tag.dir/_ieee802_11_swig_swig_tag.cpp.o
Generating moving_average_cc_impl.cc, moving_average_ff_impl.cc
arm-oe-linux-gnueabi-g++: error: unrecognized command line option '-msse2'
swig/CMakeFiles/_ieee802_11_swig_swig_tag.dir/build.make:54: recipe for target 'swig/CMakeFiles/_ieee802_11_swig_swig_tag.dir/_ieee802_11_swig_swig_tag.cpp.o' failed
make[2]: *** [swig/CMakeFiles/_ieee802_11_swig_swig_tag.dir/_ieee802_11_swig_swig_tag.cpp.o] Error 1
CMakeFiles/Makefile2:278: recipe for target 'swig/CMakeFiles/_ieee802_11_swig_swig_tag.dir/all' failed
make[1]: *** [swig/CMakeFiles/_ieee802_11_swig_swig_tag.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 14%] Built target ieee802_11_generated_sources
Makefile:123: recipe for target 'all' failed
make: *** [all] Error 2`

How can i handle with this problem? 
Best regards.

bastibl, 2019-03-29:
I'm not sure why this happens. Actually, the code first checks if `-msse2` is supported, but then the compiler fails...

It should work by deleting this part:
https://github.com/bastibl/gr-ieee802-11/blob/next/lib/CMakeLists.txt#L89-L95

Would be great if you could help debugging this, since I don't have an OpenEmbedded setup at the moment. Any chances you ran cmake with other options before and had that value cached?

rcetin, 2019-03-29:
Actually, i am trying to compile master branch and in master branch these lines are already removed from CMakeLists.txt.

https://github.com/bastibl/gr-ieee802-11/blob/08787e1ddb706b3aa40ed9170d1ab37289d876f4/lib/CMakeLists.txt#L85-L91

bastibl, 2019-03-29:
Very strange. As far as I see, I never use SSE in master.  Either you mixed up branches (since next is the default branch) or this compile switch is set  by some GNU Radio cmake (I don't think so) or I'm missing something. Please make sure that you are really on master and start the build from scratch by deleting the build directory.

rcetin, 2019-04-03:
Hi Bastian,

I guess i made something wrong. Maybe i tried last build without sourcing environment variables. Now, i tried building gr-foo again and i got this error on cmake.

rcetin@rcetin-W540:~/e300/src/gr-foo/build [±|master ✓|] $ cmake -DCMAKE_INSTALL_PREFIX=/usr ../
-- The CXX compiler identification is GNU 4.9.2
-- The C compiler identification is GNU 4.9.2
-- Check for working CXX compiler: /home/rcetin/e300/sysroots/x86_64-oesdk-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++
-- Check for working CXX compiler: /home/rcetin/e300/sysroots/x86_64-oesdk-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working C compiler: /home/rcetin/e300/sysroots/x86_64-oesdk-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-gcc
-- Check for working C compiler: /home/rcetin/e300/sysroots/x86_64-oesdk-linux/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /home/rcetin/e300/sysroots/x86_64-oesdk-linux/usr/bin/pkg-config (found version "0.28") 
-- checking for module 'uhd'
--   found uhd, version 3.14.0.HEAD-0-g6875d061
-- Found UHD: /home/rcetin/e300/sysroots/armv7ahf-vfp-neon-oe-linux-gnueabi/usr/lib/libuhd.so  
-- UHD library found.
-- Boost version: 1.57.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
Checking for GNU Radio Module: RUNTIME
-- checking for module 'gnuradio-runtime'
--   found gnuradio-runtime, version 3.7.13.4
 * INCLUDES=/home/rcetin/e300/sysroots/armv7ahf-vfp-neon-oe-linux-gnueabi/usr/include
 * LIBS=/home/rcetin/e300/sysroots/armv7ahf-vfp-neon-oe-linux-gnueabi/usr/lib/libgnuradio-runtime.so;/home/rcetin/e300/sysroots/armv7ahf-vfp-neon-oe-linux-gnueabi/usr/lib/libgnuradio-pmt.so;GNURADIO_RUNTIME_LIBRARIES_log4cpp-NOTFOUND
-- Could NOT find GNURADIO_RUNTIME (missing:  GNURADIO_RUNTIME_LIBRARIES) 
GNURADIO_RUNTIME_FOUND = FALSE
CMake Error at /home/rcetin/e300/sysroots/armv7ahf-vfp-neon-oe-linux-gnueabi/usr/lib/cmake/gnuradio/GnuradioConfig.cmake:114 (message):
  Required GNU Radio Component: RUNTIME missing!
Call Stack (most recent call first):
  /home/rcetin/e300/sysroots/armv7ahf-vfp-neon-oe-linux-gnueabi/usr/lib/cmake/gnuradio/GnuradioConfig.cmake:121 (GR_MODULE)
  CMakeLists.txt:100 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/rcetin/e300/src/gr-foo/build/CMakeFiles/CMakeOutput.log".

It cannot find log4cpp library. I guess this library is not included in toolchain. So, should i build log4cpp and put in into my toolchain?

bastibl, 2019-04-03:
The error that you experience is, AFAIS, also not related to this module. It looks like GNU Radio was built with log4cpp support (which is optional on maint) but the library cannot be found by cmake.

If log4cpp is available on your system, you can set it manually with cmake. Other than that, I'm afraid, I cannot help you much here. That would either be a question for the maintainers of the toolchain or log4cpp.

rcetin, 2019-04-05:
Hi Bastian,

I have compiled log4cpp inside of device. (Not cross compiled) So toolchain problem is solved. I also can compile gr-foo. Again, while compiling gr-ieee i have got -msse2 error. Then i edited CmakeLists.txt; (Commented out msse2 line)

```
########################################################################
# Compiler specific setup
########################################################################
if(NOT WIN32)
    #http://gcc.gnu.org/wiki/Visibility
    add_definitions(-fvisibility=hidden)
    add_definitions(-std=c++0x)
    #add_definitions(-msse2)
    add_definitions(-mfpu=neon)
    add_definitions(-mfloat-abi=hard)
endif()
```

Then i overcome msse2 error. This time it gave that error;
```
In file included from /home/root/gr-ieee802-11/lib/decode_mac.cc:20:0:
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:20:23: fatal error: xmmintrin.h: No such file or directory
 #include <xmmintrin.h>
                       ^
compilation terminated.
```

Then, i commented out this include in "/home/root/gr-ieee802-11/lib/viterbi_decoder.h". After that i got this error;

```
[ 34%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/frame_equalizer_impl.cc.o
In file included from /home/root/gr-ieee802-11/lib/decode_mac.cc:20:0:
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:56:3: error: '__m128i' does not name a type
   __m128i v[2];
   ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:59:2: error: '__m128i' does not name a type
  __m128i d_metric0[4] __attribute__ ((aligned(16)));
  ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:60:2: error: '__m128i' does not name a type
  __m128i d_metric1[4] __attribute__ ((aligned(16)));
  ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:61:2: error: '__m128i' does not name a type
  __m128i d_path0[4] __attribute__ ((aligned(16)));
  ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:62:2: error: '__m128i' does not name a type
  __m128i d_path1[4] __attribute__ ((aligned(16)));
  ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:82:4: error: '__m128i' has not been declared
    __m128i m0[], __m128i m1[], __m128i p0[], __m128i p1[]);
    ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:82:18: error: '__m128i' has not been declared
    __m128i m0[], __m128i m1[], __m128i p0[], __m128i p1[]);
                  ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:82:32: error: '__m128i' has not been declared
    __m128i m0[], __m128i m1[], __m128i p0[], __m128i p1[]);
                                ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:82:46: error: '__m128i' has not been declared
    __m128i m0[], __m128i m1[], __m128i p0[], __m128i p1[]);
                                              ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:83:40: error: '__m128i' has not been declared
  unsigned char viterbi_get_output_sse2(__m128i *mm0,
                                        ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:84:4: error: '__m128i' has not been declared
    __m128i *pp0, int ntraceback, unsigned char *outbuf);
    ^
In file included from /home/root/gr-ieee802-11/lib/frame_equalizer_impl.h:24:0,
                 from /home/root/gr-ieee802-11/lib/frame_equalizer_impl.cc:18:
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:56:3: error: '__m128i' does not name a type
   __m128i v[2];
   ^
[... 22 lines cut ...]
    __m128i m0[], __m128i m1[], __m128i p0[], __m128i p1[]);
                                              ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:83:40: error: '__m128i' has not been declared
  unsigned char viterbi_get_output_sse2(__m128i *mm0,
                                        ^
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:84:4: error: '__m128i' has not been declared
    __m128i *pp0, int ntraceback, unsigned char *outbuf);
    ^
lib/CMakeFiles/gnuradio-ieee802_11.dir/build.make:164: recipe for target 'lib/CMakeFiles/gnuradio-ieee802_11.dir/decode_mac.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/decode_mac.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
lib/CMakeFiles/gnuradio-ieee802_11.dir/build.make:325: recipe for target 'lib/CMakeFiles/gnuradio-ieee802_11.dir/frame_equalizer_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/frame_equalizer_impl.cc.o] Error 1
CMakeFiles/Makefile2:153: recipe for target 'lib/CMakeFiles/gnuradio-ieee802_11.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/all] Error 2
Makefile:123: recipe for target 'all' failed
make: *** [all] Error 2
```

So, i guess it is still related with msse instructions. There is post in here; https://stackoverflow.com/questions/11228855/header-files-for-x86-simd-intrinsics/22291538#22291538

According to it, i can include `#include <arm_neon.h>` instead of `#include <xmmintrin.h>` for ARM. But after that still same error exist. 
Do you have an idea about new updates and problem?

Regards.

bastibl, 2019-04-06:
Arg, I forgot that the master branch includes only the SSE2 implementation. AFAIS, there is no easy way to convert that to Neon. So you'd need to use the generic C implementation on ARM or put some work into converting the implementation.
Alternatively, you could either backport the SSE2/generic switch from next to master (and make  pull request ;-)) or use a GNU Radio branch that works with the next branch (the old GNU Radio next branch from a year ago or so).

rcetin, 2019-04-09:
When i remove msse2 it still gave error 
`In file included from /home/root/gr-ieee802-11/lib/decode_mac.cc:20:0:
/home/root/gr-ieee802-11/lib/viterbi_decoder.h:55:3: error: '__m128i' does not name a type
   __m128i v[2];
   ^
`

So, how can i compile for generic C implementation?

bastibl, 2019-04-09:
Yes, that's expected. Please, see the next branch for the C implementation.

rcetin, 2019-04-09:
I have changed lib/CMakeLists.txt and CMakeLists.txt according to next branch. After that i could compile master branch. But this time i got this error. 

```Traceback (most recent call last):                                    
  File "./wifi_transceiver_e310.py", line 21, in <module>   
    from wifi_phy_hier import wifi_phy_hier  # grc-generated hier_block   
  File "/home/root/.grc_gnuradio/wifi_phy_hier.py", line 14, in <module> 
    import ieee802_11     
ImportError: No module named ieee802_11
```

I have checked ieee802_11 is installed python site packages. Do you have any idea?
Thanks

bastibl, 2019-04-09:
If the module is installed and your PYTHONPATH is correct, it might be a problem of the linker.
Sometimes running the module from a Python shell shows more information. If not,  please have a look at this thread:
https://github.com/bastibl/gr-ieee802-11/issues/74

rcetin, 2019-04-09:
I installed gr-ieee to custom path via `make install DESTDIR=/home/root/localinstall/`. After that, i run `ldconfig`. 

This is my PYTHONPATH;
```root@ettus-e3xx-sg3:~# echo $PYTHONPATH      
/home/root/localinstall/usr/lib/python2.7/site-packages:
```
This is python import result;
```
root@ettus-e3xx-sg3:~# python
Python 2.7.9 (default, Jan  7 2016, 06:54:52) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import foo
>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named ieee802_11
```
I can import foo but cannot ieee802_11. Are there any difference between them?

bastibl, 2019-04-09:
The names of the installation directories can vary between platforms (site-package vs dist-packages). So, again, please check if the module is actually at that particular path. (Also I don't think that it will be under `usr`).

If the module is really in the directory that you set with the Python path and it still doe not work, it is probably a linker problem. In that case, please read the issue that I linked. Swig hides import errors and the issue should show how you can print and debug them...

rcetin, 2019-04-18:
Hi Bastian,
I have achieved to compile master branch using cross compiler. I have edited, gr-ieee802_11/CMakeLists.txt and gr-ieee802_11/lib/CMakeLists.txt for enabling generic C implementation. 
And copy some Viterbidecoder source files from next branch to master. Also, edited couple of header files then i could compile. 
Thank you for your support. 
Regards.

MarlonBedoya, 2021-01-27:
> Hi Bastian,
> I have achieved to compile master branch using cross compiler. I have edited, gr-ieee802_11/CMakeLists.txt and gr-ieee802_11/lib/CMakeLists.txt for enabling generic C implementation.
> And [...]

Hello rcetin. I have the same problem as you. I wonder if you can help me with the process you made to replicate it.