bastibl/gr-ieee802-11 #122: Add parsed data as parse_mac output

creffett, 2018-07-09:
Hi! As part of a project I'm working on, I'm going to add a message output to parse_mac to output a message with the parsed data every time a packet comes into the block (so it would be a message with packet type, SSID, MACs, etc.). Is this something you'd be interested in? If so, I'll open a PR sometime this week when I've made the changes.

bastibl, 2018-07-10:
Hi, that depends if this is a trivial extension for your experiment or if that's generally useful and nice to have. Is there an interesting use-case? If one wants to parse the payload, why not feed data to Scapy or use the Wireshark Connector and parse the PCAP with tshark, for example.

creffett, 2018-07-10:
In our case, we're trying to cut down on extra components and just have the flowgraph output the data extracted from the headers from packets as they come in instead of having to pass the data through tshark or scapy (and then the data is going out via a ZeroMQ sink).

bastibl, 2018-07-10:
To be honest, I don't like the Parse MAC block, since it parses only a small subset of frames. It was mainly for debugging during development.
Maybe a nicer way would be to forward the whole frame and parse the fields of interest at the other end (if it's really an issue to parse the data locally with Scapy or tshark). I hope you don't mind if I don't add this. I think it would be, in any case, only a random subset of the fields that would get forwarded and everybody would have to adapt it to his or her needs.

creffett, 2018-07-10:
Sure, no problem, that's why I asked. Closing.

closed 2018-07-10 by creffett