bastibl/gr-ieee802-11 #124: Can't Receive Some Types of Frames

ggamble22, 2018-07-18:
Hi @bastibl, I am running into a problem where I can't receive beacon or eapol handshake frames. I am operating on a 2.4GHz network and it is being used in an isolated room so no other signals are present. I am seeing other frames like acknowledgment frames. 
My current setup:
Ubuntu  18.04
gnuradio 3.7.12
gr-ieee latest master branch

I have tried adjusting the setting the available settings like gain. I know the handshake might be different so I just want to make sure there isn't anything that may filter out frames that could be a slightly different protocol. In Wireshark they show up as using the EAPOL protocol rather than 802.11. Thanks for your help.

bastibl, 2018-07-18:
How do you know that they are not received. Any chances you just look at the output of the Decode MAC block, which parses only a small subset of frames. Did you check the PCAP file with Wireshark?

ggamble22, 2018-07-18:
Sorry, I left out my test setup. I have two instances of Wireshark running. One is with the USRP example and the other is with my Wi-Fi adapter in monitor mode. For the beacon frames, I have found that I can see them from other networks, just not then ones from the Phantom so it may be sending theem in the legacy B mode.

bastibl, 2018-07-19:
if that's the case for beacons that might also be the case for other management frames. I would recommend to switch to 5GHz. There is no 11b mode, or 11b-compatibility mode.

ggamble22, 2018-07-19:
Unfortunately I can not change the frequency used by the drone. Out of curiosity, why isn't the 11b mode supported and what would it take to implement it.

bastibl, 2018-07-19:
11b is a DSSS PHY. 11a/g/p is OFDM. That's totally different. There was an 11b implementation for GNU Radio, but I think it is no longer maintained and needed some HW acceleration. 

See:
T. Vilches and D. Dujovne, “GNURadio and 802.11: Performance Evaluation
and Limitations,” IEEE Network, vol. 28, no. 5, 27–31, Sep. 2014. DOI:
10.1109/mnet.2014.6915436.

ggamble22, 2018-07-20:
Ok, Thanks.

closed 2018-07-20 by ggamble22