bastibl/gr-ieee802-11 #128: Fatal error when trying to build CMake Files

Nodemcu-ife, 2018-09-14:
Following the instruction on how to install the dependency for gr-foo i ran into fatal error. May i know what possible way to solve this. However, the gr-ieee802-11 was able to install succefully but with blocks missing because i was unable to install gr-foo successfully. I tried downloading the gr-foo from your git hub repository and still couldn't successfully install. Please, I would appreciate if you could guide me on how to fix this issue. Thanks. My version of GNU Radio is GNU Radio Companion 3.7.10.1 running on Ubuntu16.04. 
The error i received fro m the terminal while trying to "make" gr-foo in the termninal is this:
 :~/gr-foo/build$ make
[  4%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/channel_model_impl.cc.o
In file included from /home/home/gr-foo/lib/channel_model_impl.cc:18:0:
/home/home/gr-foo/lib/channel_model_impl.h:27:47: fatal error: gnuradio/filter/mmse_resampler_cc.h: No such file or directory
compilation terminated.
lib/CMakeFiles/gnuradio-foo.dir/build.make:86: recipe for target 'lib/CMakeFiles/gnuradio-foo.dir/channel_model_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-foo.dir/channel_model_impl.cc.o] Error 1
CMakeFiles/Makefile2:135: recipe for target 'lib/CMakeFiles/gnuradio-foo.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-foo.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

bastibl, 2018-09-15:
I guess you are using the wrong branch for your GNU Radio installation. See the readme.

Nodemcu-ife, 2018-09-15:
I installed the GNU Radio software from Ettus installation page:https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux
May I know if i have to uninstall the current GNU Radio i have now and install the one you prescribe in your readme.
Thanks

bastibl, 2018-09-16:
I don't know about this image/toolchain. As I said, you probably use the wrong branch of gr-foo. It has only two branches. Couldn't you just try to compile the other one?

closed 2018-09-28 by bastibl