bastibl/gr-ieee802-11 #129: simulating without usrp

Nodemcu-ife, 2018-09-27:
Hi,
I'm sorry to inconvenient you. May I know how to include a noise channel and make a simulation that runs in gnu radio only without the need of the usrp. Can you direct me what to remove and what to add.
Thanks

bastibl, 2018-09-28:
That's what the loopback flow graph in the examples directory does.

Nodemcu-ife, 2018-09-28:
Thank you for showing me to the right direction. 
Furthermore, May i know if i need to redo the cmake because as i decrease the snr the error rate does not respond it just stays at 0.00%.

bastibl, 2018-09-28:
No, that's not compiled code. The controls are interactive, but the displayed error rate is just a very simple implementation that requires that at least some frames get through. Do not use it for proper performance evaluation.

Nodemcu-ife, 2018-09-28:
Ok, sir. May I know how i can get configure the flow graph to get error rate in AWGN noise source.

bastibl, 2018-09-28:
See the AWGN simulations in the simulations folder.

closed 2018-09-28 by bastibl

Nodemcu-ife, 2018-09-28:
I'm sorry i get this error while trying to run the graph. Thanks

linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.001.HEAD-0-g929e3b32

MAPPER: encoding: 0
set_min_output_buffer on block 9 to 397056
set_min_output_buffer on block 11 to 397056
set_min_output_buffer on block 13 to 397056
set_min_output_buffer on block 14 to 397056
set_min_output_buffer on block 16 to 397056
set_min_output_buffer on block 28 to 96000
thread[thread-per-block[1]: <block random_periodic_msg_source (27)>]: pmt_to_long: wrong_type : #t

Nodemcu-ife, 2018-09-28:
I've spend some time looking at how the sim.grc flow graph works while also keeping an eye on the wifi_phy_hier block. I am running the No GUI sim.grc and changing the noise amplitude does not show any responds both in my output terminal and file. the output from terminal is below here:
MAPPER: encoding: 0
set_min_output_buffer on block 9 to 397056
set_min_output_buffer on block 11 to 397056
set_min_output_buffer on block 13 to 397056
set_min_output_buffer on block 14 to 397056
set_min_output_buffer on block 16 to 397056
set_min_output_buffer on block 28 to 96000
thread[thread-per-block[1]: <block random_periodic_msg_source (27)>]: pmt_to_long: wrong_type : #t
May i know what i'm missing. I'm sorry for asking this small petty questions but I ma rather left confused,sir.

Nodemcu-ife, 2018-09-28:
I am sorry for posting multiple times. Indeed I hope you can forgive my intolerable altitude. Thank you. I will remake the gr-foo and update you on that issue.