bastibl/gr-ieee802-11 #132: awgn simulation

Nodemcu-ife, 2018-09-28:
I've spend some time looking at how the sim.grc flow graph works while also keeping an eye on the wifi_phy_hier block. I am running the No GUI sim.grc and changing the noise amplitude does not show any responds both in my output terminal and file. the output from terminal is below here:
MAPPER: encoding: 0
set_min_output_buffer on block 9 to 397056
set_min_output_buffer on block 11 to 397056
set_min_output_buffer on block 13 to 397056
set_min_output_buffer on block 14 to 397056
set_min_output_buffer on block 16 to 397056
set_min_output_buffer on block 28 to 96000
thread[thread-per-block[1]: <block random_periodic_msg_source (27)>]: pmt_to_long: wrong_type : #t
May i know what i'm missing. I'm sorry for asking this small petty questions but I ma rather left confused,sir.

Nodemcu-ife, 2018-09-28:
Again sir, 
I am sorry for r-posting this issue here again. Thank you for your kindness. I highly appreciate.

Nodemcu-ife, 2018-09-29:
Hi Bastian Bloessl, I hope i can ask you a question or two.
May i know how to calculate SNR from varying the noise source in awgn simulation. thank you

Nodemcu-ife, 2018-09-29:
I merged master and next together and re-run the sim.grc for awgn simulation and am still getting error like this
[image]
 regarding pcap. Can you advice on what i should do next.

bastibl, 2018-09-29:
This is a simulation. You do not calculate the SNR but set it. If you'd look at the flow graph you see that the SNR is a parameter. You can set it on the command line.
Please just look at the flow graph it should be obvious. (In a nutshell: noise is set to power 1, the PHY outputs the signal with an output power of 1, the amplitude of the signal is adjusted in the Multiply Const block to the target SNR.)

You said you merged master and next. That's not how software development works. You are not supposed to merge branches but use either of them. (When you merge Linux 2 and Linux 4, you neither get Linux 3 nor Linux 7. You get a mess. Same thing here.)

The simulation is supposed to be run through the Makefile. It automates the whole process of running, parsing, and plotting). Please google what a Makefile is, what it is good for, how to use it, and how it is used in this simulation.