bastibl/gr-ieee802-11 #135: Different Channel Profiles in Single Tuning

usmanwardag, 2018-10-28:
Hello Bastian.

I am extracting the channel phase response for all the sub-carriers, but I am a little confused about the output I am getting. I will briefly explain my setup and the way I am extracting d_H.

I am using USRP N210 with a wired connection between TX and RX ports (hence no channel noise). Messages are continuously transmitted using BPSK 1/2 modulation and decoded using LS estimator. To get the channel response I print out `d_H` in `ls.cc` for n=4. 

Now, when I plot the phase response for over 1000 packets, I get 5 distinct phase profiles. Please see the figure in image [here](https://ibb.co/jFYL5q). The X-axis has 51 sub-carriers (I have removed the guard bands and central sub-carrier) and the Y-axis has the phase in radians. 

All the packets are collected in a single USRP tuning, so I am a little confused why I am seeing 5 different phase profiles. Is there a reason that could explain the results I am seeing? 

Thank you very much for your time!

usmanwardag, 2018-10-28:
Thanks for the response!

It constantly changes back and forth. I have also tested over a wireless channel and with different sampling rates but I still see the same problem. Is there some offset estimation going on in the code that could explain this? 

As I understand, the LS estimator estimates channel for a few training symbols. If I were to increase the number of training symbols, would it help solve the problem? Finally, I am observing a Gaussian distribution for the subcarrier phase across a large number of packets. So, if I average the channel response across a large number of packets, would the mean response be a decent representation of the channel?

bastibl, 2018-10-29:
Another thing that would come to my mind is sampling frequency correction in the frame equalizer. Can you try disabling the `d_epsilon0` and `d_er` stuff in `lib/frame_equalizer.cc`.

closed 2019-02-18 by bastibl

sonebu, 2023-05-13:
Hi, sorry to bump an old thread, but this potentially-unsolved issue seemed important for what I'm currently trying to understand in my own project (see _[note]_ below for details). @bastibl I can remove this post or move the discussion elsewhere if you don't think this is the appropriate place.

@usmanwardag @bastibl:
- did you end up discovering why this happened? If not, and if it's OK for you, it would be great to hypothesize on it a bit together
- (the link to the original image seems broken, so, sorry in advance if this was already visible there) trying to understand the phase profile definition --> I'm assuming 1000 packets and 51 subcarriers means we basically have 1000 curves of 51 datapoints each, and you saw that these 1000 curves were grouped into 5 distinct "shapes" (X: subcarrier index, Y: 1 phase value for that subcarrier). I guess the expected shape for each curve is a pretty straight (almost horizontal?) line, you mentioned a Gaussian distribution here so I'm feeling I might have missed something here. Did these 5 groups simply have offsets in the Y axis between them, or did they also have different shapes? Attaching a picture that hopefully depicts the two cases I'm referring to. 

[image]

If it's Case A...
- What was this wired loopback setup like, electrically (i.e., how long was the cable, did it have a large in-line attenuator etc., how much was the gain)? This is just a wild guess, but if the setup was transmitting and receiving simultaneously, [the TX RX isolation on the N210 seems to be around 30dB](https://forums.ni.com/t5/USRP-Software-Radio/How-to-isolate-the-signals-between-TX-RX-and-RX2-on-the-USRP/td-p/2267190), so I wondered if a TX signal leakage on your RX end was causing this weird behavior (different leakage paths would have different lengths, and since the wavelength is very short for these 802.11 bands, the different paths could appear like multipath components and cause this). Still, if the 5 phase profile groups you saw were very distinct, this does not concretely explain why there were exactly 5.
- by "It constantly changes back and forth", were you getting kind of random 5 phase profiles every time you analyzed a batch of 1000 packets or was there a certain pattern (like 2 states for "back" and "forth")?

I don't have a guess for Case B.

Eager to hear your takes on this.

_[note] I'm hoping to develop a precise object detection-localization system based on 802.11 CSI info. Currently trying to understand how precise the channel estimation algorithms can get before hitting some fundamental (algorithm performance against noise etc.) and/or practical (radio chain, hardware-related etc.) barriers. If I'm understanding @usmanwardag s test setup correctly this is a pretty good "toy" setup where you would expect most things to work nearly perfectly so understanding the phenomenon he experienced would be informative._

P.S. @bastibl thanks for this package and your contributions to this field in general, I've learned a lot from all of it.

bastibl, 2023-05-15:
Thanks for following up on the topic. Your argument for case A sounds very reasonable to me, but I don't have any measured data available. Maybe @usmanwardag could share some?