bastibl/gr-ieee802-11 #137: Calculating Packet Error Rate

Nodemcu-ife, 2018-11-09:
Hi Bastian,
I successfully made the tx and rx work on usrp n210. however I want to be able to calculate Packet error rate or bit error rate. Thus may i know what adjustment can be made to the flowgraph in order to achieve this. Or whether I had to calculate it manually by sending binary file.
Thanks again.

bastibl, 2018-11-09:
There is an example for PER simulations in the simulations folder. It can be easily adapted for over-the-air measurements. Biggest problem might be that the latest GNU Radio release has issues with run to completion flow graphs (which are used for the simulations). If you have any issues try the `next` branch of this module with the `next` branch of my GNU Radio mirror (https://github.com/bastibl/gnuradio).

Nodemcu-ife, 2018-11-09:
Thank you.
if i may ask, How do can i calculate packet error rate for ofdm example gnuradio ?

bastibl, 2018-11-09:
If you send `s` packets and receive `r`, then `per = (s-r)/r`.
If you want to know how to setup and automate simulations for the GNU Radio OFDM transceiver, please write to the GNU Radio mailing list.

Nodemcu-ife, 2018-11-09:
Thanks, for the calculation tips
But I am asking in the case where i am sending over the air. 
thanks

closed 2019-02-18 by bastibl