bastibl/gr-ieee802-11 #138: gr-ieee802.11n Adalm Pluto

gorlapraveen, 2018-11-11:
Can this be used with Adlam pluto 
1. for establishing WiFi connection with Mobile Devices 
2. or with other SDR Adlam devices ove ieee802.11n

bastibl, 2018-11-11:
This is mainly a PHY implementation of 802.11a/g/p. It doesn't support 11n features. It can send/receive frames frames to/from WLAN cards but it doesn't support all signaling out of the box. That means when you start the GNU Radio flow graph you won't see another WiFi card in your PC. Also, implementing the PHY on the PC implies that we cannot meet the timing constraints of the MAC (ACK and RTS/CTS timings).

gorlapraveen, 2018-11-11:
@bastibl Thanks!. Ok,  means that it can be used with GNUradio along with Adlam Pluot kind of SDR for 802.11a/g  implementation. If then, can it be able to connect Smart phone or other PC  just for exchange of Frames, in order to get other devices( Smartphone/PC) MAC and Signal strength kind of things?
Thanks in advance!

bastibl, 2018-11-12:
> can it be able to connect Smart phone or other PC just for exchange of Frames, in order to get other devices( Smartphone/PC) MAC and Signal strength kind of things?

Again, there is no signalling implemented out of the box. So it will not be able to "connect" to the networks (associate, get IP, etc). But of course you can receive frames and estimate signal quality. But since the SDR is no calibrated measurement device, you won't get an absolute signal strength out of it.

You do not have to "bind" to a device to receive frames. Frames are sent over the air and can be received by any device.

gorlapraveen, 2018-11-12:
> You do not have to "bind" to a device to receive frames. Frames are sent over the air and can be received by any device.

@bastibl Thank you!.
So  With SDR, I can receive WiFI frames transmitted by any other devices such as Smartphone, PC and SDR with both MAC and RSSI(not absolute).

bastibl, 2018-11-12:
Yes, like with a WLAN card in monitor mode, you can receive frames of other devices (if they are 11a/g/p).

gorlapraveen, 2018-11-14:
@bastibl , But using RFTap and Wireshark , the Signal Power is not obtained, it is showing zero as seen in the below picture. Please help in this reqard.

 Thank you!
[image]

gorlapraveen, 2018-11-14:
@bastibl Can that PCAP file contains Signal Power as well along with SNR?

Thank you!

bastibl, 2018-11-14:
It can, but it doesn't out of the box. As I already mentioned, the SDR cannot determine absolute signal power. You can add something like dBFS if you want. Please have a look how SNR is implemented and add power accordingly.

gorlapraveen, 2018-11-14:
OK, Thankyou! @bastibl

gorlapraveen, 2018-11-15:
[image]

@bastibl  Here Radiohead is hsowing SSI signal and SSI Noise, Does SSI signal is nearly equal to RSSI ?
Thank you!

bastibl, 2018-11-16:
Hard to tell. I don't think that there is a strict definition of RSSI. It's just some random number that gets bigger when the signal quality gets better. Also, when there is interference, the signal power might get higher while the signal quality gets worse. So it all depends on what you actually need those number for....

That being said, at the moment, the receiver only estimates the SNR and sets the noise and signal field in the radiotap header in way that the SNR is as estimated.

https://github.com/bastibl/gr-foo/blob/next/lib/wireshark_connector_impl.cc#L94-L111

If you want the signal power, you'd have to propagate it with a tag, similar to how SNR is propagated.

gorlapraveen, 2018-11-19:
@bastibl  As we can find `fer`(Frame Error Rate), Can you help in knowing that how can `FER` help in correcting the Signal Strength, based on instantaneous FER.

Thank you!

bastibl, 2018-11-19:
Sorry, but I have no idea what you are trying to say or ask.

gorlapraveen, 2018-11-19:
@bastibl . What I am saying is that, As we were able to get `Signal Strength` and `fer(Frame Error Rate)`. As `Signal Strenght` is not absolute(it may be inaccurate),  So can you please say that how can we use `fer` to correct `Signal Strength`.

bastibl, 2018-11-19:
I still don't fully get it, but I'm afraid that's not possible or doesn't make any sense. Signal strength is not absolute but that doesn't mean that it is inaccurate. It just means that there is a constant, unknown shift between the measured value and the absolute value.

No idea what you are trying to plot or find out. But in any way, if you wanted to use the FER "correct" the signal strength you'd need a frame error rate model or something. But again, I believe that's just the wrong approach...

closed 2019-02-18 by bastibl