bastibl/gr-ieee802-11 #139: Problem with signal strength in wireshark

wshe1, 2018-11-20:
Hello Bastian，I am using your tool  with a usrp b210 and want to get the signal strength.The signal strength is generally negative, but the signal strength value I get from wireshark is positive.I have been confused about it, so I would like to ask you how to turn it into a negative value, or the value represents other meanings.
[image]

bastibl, 2018-11-20:
The USRP is no calibrated measurement device and, therefore, cannot measure absolute power. The value in Wireshark is set so that the SNR corresponds to the estimated SNR. (For the frame in the picture it would be an SNR of 30db).

bastibl, 2018-11-20:
It's 31dB because the noise is 0. Wireshark doesn't do any magic. It's set in the Wireshark Connector block
https://github.com/bastibl/gr-foo/blob/next/lib/wireshark_connector_impl.cc#L94-L111

If you want the amplitude of the signal you have to take it from Sync Short and propagate it through the flow graph like the frequency offset.

closed 2019-02-18 by bastibl