bastibl/gr-ieee802-11 #144: RuntimeError: Failed to create FFTW wisdom lockfile

madatr, 2018-12-04:
Hello,

I am trying to use the LimeSDR-mini with gnu radio to capture wifi data.

My set up is:
LimeSDR-mini
OS Ubuntu 18.04.1
Downloaded Gnuradio using pybombs.
Downloaded gr-foo, gr-ieee802-11
Trying to run wifi_loopback.grc results in the following:

`
Executing: /usr/bin/python2 -u /home/user/gr-ieee802-11/examples/wifi_loopback.py
Gtk-Message: 18:20:14.245: Failed to load module "canberra-gtk-module"
MAPPER: encoding: 0
set_min_output_buffer on block 9 to 397056

Traceback (most recent call last):
  File "/home/user/gr-ieee802-11/examples/wifi_loopback.py", line 350, in <module>
    main()
  File "/home/user/gr-ieee802-11/examples/wifi_loopback.py", line 338, in main
    tb = top_block_cls()
  File "/home/user/gr-ieee802-11/examples/wifi_loopback.py", line 149, in __init__
    sensitivity=0.56,
  File "/home/user/.grc_gnuradio/wifi_phy_hier.py", line 59, in __init__
    self.fft_vxx_0_1 = fft.fft_vcc(64, True, (window.rectangular(64)), True, 1)

  File "/home/user/prefix/lib/python2.7/dist-packages/gnuradio/fft/fft_swig.py", line 154, in make
    return _fft_swig.fft_vcc_make(fft_size, forward, window, shift, nthreads)

RuntimeError: Failed to create FFTW wisdom lockfile: /home/user/.gr_fftw_wisdom.lock
>>> Done
` 
I have tried looking on the interweb and could not find any similar issues.

bastibl, 2018-12-04:
I close this since you cross-posted to the GNU Radio repo.

closed 2018-12-04 by bastibl

bastibl, 2018-12-04:
Opening several issues without linking them leads to independent discussions, wasting everybody's time.