bastibl/gr-ieee802-11 #145: Estimate of the USRP TX power

wshe1, 2018-12-11:
Hello Bastian:
I am using USRP B210 and want to estimate the he transmit power. However, I don't know how to estimate the transmit power. I sincerely hope that you can give an example of estimating the transmit power.

wshe1, 2018-12-11:
Hello Bastian:
   Based on your prompts, I  rough estimate the transmitter output power as follows:
   At a frequency of 5900Mhz, I set the transmitter's gain tx_gain=1.00. 
[image]

Then, based on the USRP B200 RF performance graph, I calculate the output power as:
X axis： 90*1.00=90（dB）
Y axix: the Power  rough estimate: 10 (dBm)
[image]

I want to know if I rough estimate outpower is correct?

bastibl, 2018-12-11:
Sorry, I can give you pointers and tips, but I cannot approve any calculations. That's your responsibility. You'll have to look into that on your own until you understand what's going on and know what you are calculating.

closed 2019-02-18 by bastibl