bastibl/gr-ieee802-11 #146: VOLK support

augustojv, 2018-12-14:
Hello Bastian.

I've been trying to measure the benefits of using GNURadio's VOLK when running the gr-ieee802-11 transceiver. However, I don't see much improvement when VOLK is enabled versus the VOLK disabled case. Does the code make any use of VOLK? If not, are you considering adding VOLK support in the future? I would like to know your thoughts about this.

Thanks!
Augusto

bastibl, 2018-12-15:
Hi Augusto,
VOLK is used in `multiply`, `complex_conjugate`, `sync_long`,  `fft`, etc, etc. While the exact numbers depend on the architecture, I'd expect a big difference. Maybe, your measurement setup was not working or you did not run `volk_profile`.

augustojv, 2018-12-16:
Thanks for your quick response, Bastian. I'm pretty sure that Volk is running fine in my platform since I do see significant performance differences (for some kernels) when I run the Volk benchmarks provided with GNU Radio (and yes, I executed `volk_profile`). Still when I run `gr-ieee802-11`, I don't get much benefit (or no benefit at all). In my case, execution time is dominated by the `WiFi Mapper` block which takes most of the time. Does this block exploit Volk? If it does not, then this may be a possible explanation.

Thanks!
Augusto

bastibl, 2019-01-19:
Until now, I didn't dig into individual function. I only looked at block-level performance.
We know that PMTs are bad and are currently looking for a replacement. I think at the Hackfest in two weeks, we'll look into alternatives. At least, it's on the agenda: https://wiki.gnuradio.org/index.php/Hackfest1902

I assume that strings are especially bad. Let me know if you can share performance measurements.

augustojv, 2019-01-21:
Thanks, Bastian -- let me try to send you some performance measurements separately.

Another weird behavior that I'm observing is that performance can change radically if I "play" with core affinity in my multi-core CPU (i.e. if a pin some GNU Radio blocks to some specific cores, or even all the blocks to one single core, versus letting the application run freely in the CPU). Have you experienced something similar? 

Thanks.
Augusto

augustojv, 2019-01-25:
Hi Bastian.

Just to update you, the problem that I mentioned before about `mapper` consuming most of the CPU was due to the busy-waiting mechanism that it uses to poll the input port (`delete_head_nowait`). I increased the rate at which I inject data into `mapper` (to keep it doing actual work), and the problem went away. Now my measurements match the ones in your paper _"Performance Assessment of IEEE 802.11p with an Open Source SDR-based Prototype"_.

Still, the fact that I don't see benefits between Volk "off" and "on" keeps spinning round in my head. You're right -- Volk is used in `multiply`, `complex_conjugate`, `sync_long`, `fft` and others. But except for `sync_long`, the other blocks represent a marginal portion of the overall execution time (as you show in Fig. 6 of your paper). And even going deeper into `sync_long`, most of its time goes to the computation of complex exponentials (which I don't think that has a Volk implementation). So, if you're getting benefits from using Volk, I'm still struggling to see what I'm doing wrong here... :)

Anyway, I'll let you know if I can find a way around it.

Thanks.
Augusto

bastibl, 2019-01-26:
Thanks for the update. The busy-wait is (or was) a problem of the GNU Radio scheduler. My understanding was that more recent versions of GNU Radio wait a bit if nothing was produced (return 0 in work).

https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/tpb_thread_body.cc#L157

Also, for the measurements in the paper, I didn't include the TX, it was only RX. So the mapper was not part of the test. In your case, it sounds like the mapper block captured one CPU core. Also if there is no CPU affinity, I could imagine that the mapper might be scheduled all the time on different cores, which might degrade performance considerably.

Regarding Volk, I had a first implementation, where I did most of the synchronization in custom blocks (i.e., multiplications, complex conjugate, averaging) and this didn't work at all. I switched to the VOLK-accelerated blocks and everything worked in real-time. How do you test Volk on/off? Maybe you test different implementations of SIMD but not SIMD on vs SIMD off. Shouldn't Volk always provide an advantage over a naive implementation, where one, for example, multiplies sample by sample in a for loop? I didn't dig deeper into how Volk works under the hood (i.e., how it selects implementations).  I can have a closer look into it if it still causes issues.

closed 2019-04-09 by bastibl