bastibl/gr-ieee802-11 #148: wifi_rx.grc enables wireshark connector and file sink

Zarados, 2019-01-02:
Hi, I try to demodulate DSRC signal to pcap file.

If I enable wireshark connector and file sink, the QT GUI sink won't open. 
(./rx_demo.sh only open wireshark)

But if i disable these blocks, the flow graph is work!

Did you have any idea?

++++++++++++
<<< Welcome to GNU Radio Companion 3.7.13.4 >>>

Block paths:
	/usr/local/share/gnuradio/grc/blocks

Loading: "/root/git/gr-ieee802-11/examples/wifi_rx.grc"
>>> Done

Generating: '/root/git/gr-ieee802-11/examples/wifi_rx.py'

Generating: '/root/git/gr-ieee802-11/examples/wifi_rx.py'

Executing: /usr/bin/python2 -u /root/git/gr-ieee802-11/examples/wifi_rx.py

[32;1m[INFO] [UHD] [39;0mlinux; GNU C++ version 8.2.0; Boost_106200; UHD_3.14.0.0-98-g0523fd94

++++++++++++

I don't know why it stop in " [32;1m[INFO] [UHD] [39;0mlinux; GNU C++ version 8.2.0; Boost_106200; UHD_3.14.0.0-98-g0523fd94".

https://imgur.com/a/JNzGWTi

If I want to stop, I need use kill the flow graph.

bastibl, 2019-01-03:
just start the rx flow graph from grc or execute the generated python script.

Zarados, 2019-01-03:
Ok, I rebuild a vm to fix it.
I have used SoapySDR simulate USRP before, I think SoapySDR makes this problem.
Anyway, thanks alot.
Should this project can demodulate DSRC signal?

bastibl, 2019-01-03:
Great, yes it can.

closed 2019-01-03 by bastibl

Zarados, 2019-01-09:
Hi, I think I met some trouble with DSRC signal.
Can you help me?

1.When i use wifi_rx.grc or rx_demo.sh, i can't demodulate DSRC signal. I don't know why.
My DSRC signal channel is 172 and the bandwidth is 8M. (I use inspectrum to check it.)
So I set channel to 172 and the sample rate set 20M.
https://imgur.com/a/LP3n7J5

2.And I record DSRC signal with osmocom_fft. 
I use file source instead of UHD source, and add two FIR filter to replace two moving average block.
But i still can't demodulate it.
https://imgur.com/a/BXM6z1O

I have to make a little change with wifi_rx.grc.
I set the default value of frequency to 172, and default value of sample rate to 20M.
Maybe I can share DSRC.cfile to your google drive, If you want to it. (It is a little large. 1.92G)

bastibl, 2019-01-09:
There is no 8MHz DSRC signal. And even if, why then set the sample rate to 20MHz... It's 10MHz. Do not change the flow graph, only select bandwidth and channel while the receiver is running (channel 172, 10MHz). 

The output is debug information, I'd recommend to disable it. See the GNU Radio documentation for logging options.

bastibl, 2019-01-24:
If you'd only use one ADC, you'd have to sample the 10MHz signal with 20MHz and get real values. The SDRs, however, use two ADCs that mix the signal with a sine and a cosine of the carrier frequency to produce complex samples with I and Q components. In that case you  sample the 10MHz signal with 10MHz and get complex values. (It's the same amount of data. Google for direct conversion if you're interested in more details.)

The signal appears to be smaller since the subcarriers at the edge are not use (guard subcarriers). But the whole structure of the frame is aligned for 10MHz, so it's much easier to process if the signal is sampled with 10MHz.