bastibl/gr-ieee802-11 #149: 802.11ah implementatin

jaaanu, 2019-01-25:
Hi,
We have been working on 802.11ah PHY implementation.
2MHz operation of 802.11ah requires 52 data subcarriers instead of 48 in 802.11a/g.
Would changing the carrier index in ofdm carrier allocator block suffice.?
Kindly guide for making proper changes..
Thanks.

bastibl, 2019-01-25:
This is a bit like asking, if I have a motor cycle and add two more wheels, do I get a car?

I don't know how exactly your PHY looks, but you'd have to change a lot. For example, how much symbols you need for a given number of data bits, the receiver would have to decode these subcarriers, some equalizers assume guard subcarriers to be zero, the interleaver pattern is designed for a given number of subcarriers, etc. etc. There's  really lot of stuff to be changed.

jaaanu, 2019-01-25:
Can i know where exactly in implementation LTF and STF fields are added at TX

jaaanu, 2019-01-25:
phy is more or less same with minor carrier parameter changes,..

jaaanu, 2019-01-25:
In the carrier allocator block all carriers are specified instead of data sub carriers alone..do we need to specify data subcarriers alone

bastibl, 2019-01-25:
This is not a chat. Please think first and then ask a question.
The training symbols are added in the carrier allocator ("sync words").
And again: minor changes to the subcarrier allocation -> many changes throughout the whole transceiver.
No idea what you mean. There are (1) total number of subcarriers, (2) data subcarriers (occupied carriers), and (3) pilot subcarriers. I guess you can figure out what is what, when looking at the carrier allocator block.

closed 2019-01-25 by jaaanu