bastibl/gr-ieee802-11 #151: Installation in MAC OS

sayaz, 2019-01-27:
Hi Bastian,
I am already using your blocks in Ubuntu OS where everything is working fine. I am not very fluent in using macOS, but it is a need for me to run your blocks in macOS. 

I have installed Gnuradio-companion, with macports as said in [here](https://wiki.gnuradio.org/index.php/MacInstall), which is also running fine. But I don't get the usual gr-xxxx folders as in Ubuntu, not even by searching them. Can you please let me know the procedure for installing this in macOS High Sierra (10.13.6 version)

sayaz, 2019-01-27:
Sorry for the trouble. You can close this issue. I saw the solution in your comments on another post as below, this has solved the problem.

[grc]
global_blocks_path = /opt/local/share/gnuradio/grc/blocks
local_blocks_path = ~/usr/share/gnuradio/grc/blocks

bastibl, 2019-01-27:
Great that it worked.

closed 2019-01-27 by bastibl