bastibl/gr-ieee802-11 #153: Regarding creating a dataset for machine learning

arun121cs, 2019-02-13:
Hello,
I am trying to use Hack RF one to send and receive back the same signal and i want to understand the differences between both the signals. I actually want to collect some data and create a machine learning model which can classify objects using RF signal. Can you please guide me how can i create a GNU radio flowgraph which gives me a file in form of a dataset which i can use for machine learning?

bastibl, 2019-02-13:
To interface the HackRF with the WLAN module, you'd have to replace the USPR source/sinks with Osmocom source/sinks  (or Scapy source/sink). (Note that the HackRF does not do any DC offset compensation. This can lead into problems, when it is not compensated for).

Regarding the ML thing: it's hard to tell what to do. It depends on what you want to feed into your ML algorithm. The samples, the samples after frame detection, the samples after synchronization, the complex constellation points, etc etc. In general, its mainly a question about which data stream to pipe into a file sink.

arun121cs, 2019-02-13:
Actually i have two main questions:-
1. Which data will be more sensitive to reflection from you mentioned?
2. Is it possible to create a GNU radio flow graph which swithes modes for tx and rx as quick so that i can capture and transmit signal at same time? 
I tried the transciver from examples using osmocom sink but i was not able to understand how to make use of it. I am actually new to signal processing but i think i can use this data for machine learning.  Your views will be valuable regarding it.

bastibl, 2019-02-14:
I think you have to dig a bit deeper into the topic and think about what it is exactly that you are planning to do. I guess with reflection you mean multi-path effects. They have an impact on all levels of the communication system. So there is no clear answer to it. It might be exactly the question that you might have to answer.

You can not send the frame and then switch to reception to receive the very same frame. What you want is full-duplex, i.e., a device that can send and receive at the same time. There are SDRs that are full-duplex capable, but I suggest that you first do some simulations with channel models (-> loopback flow graph) before you do any over-the-air stuff. When things work as expected, you can switch to hardware. Ideally to two separate SDRs that are spaced a bit apart from each other.

closed 2019-04-09 by bastibl