bastibl/gr-ieee802-11 #155: Extract the 802.11a/g PHY layer only and experiments with a synced link

tingjunchen, 2019-02-16:
Hi Bastian,

Thank you for having this wifi framework implementation and I am able to both run the txrx_loopback (in simulation) and use two separate USRPs for Tx and Rx. I have the following 3 questions:

Q1. For our research, we would like to mainly evaluate our system in the PHY layer, so it would be really helpful for us to only use the PHY modules with different MCSs that are wifi-compatible. I understand that the wifi_tx flowgraph can be used to generate the Tx frames. For the Rx, if we don't care about the frame detection (at least at this point) and would still like to have the rest of the Rx functionalities (freq offset correction, demodulation/decoding), how would it work? 

I have tried to have 2 USRPs synced through a MIMO cable and start the Tx and Rx flowgraphs on them at the same time. In particular, since the Rx samples are synced with the Tx samples through MIMO cable, I feed the Rx stream directly into the sync_long block but it doesn't seem to work... Any suggestions on this?

Q2: I also observed both late packets (LLLL) and underflows (UUUU) depending on the PDU length and the interval of the message strobe block. Do you have any suggestions on this?

Q3: Would the same PHY layer (the 8 MCSs you have implemented) also work for other frequencies (e.g., sub-2GHz cellular frequencies)?

Thank you in advance for your time and help!
Tingjun

bastibl, 2019-02-16:
Q1: Sorry, I don't understand. You want to receive frames, but you want to remove the frame detection module? Even if you don't care about frame detection, I'd recommend to just leave it in the flow graph.

Q2: You have to lower the number of packets that are sent, i.e., increase the interval in the Message Source block. The reason can be any of:
- the frames are not created fast enough (the GNU Radio blocks for creating frames are very very inefficient. )
- the bus to the SDR is not fast enough
- the rate is beyond physical limitation (the generation interval is shorter than the frame time)

Q3: Yes, just change the frequency.

tingjunchen, 2019-02-17:
Q1: My apologies if my questions were not clear... In order to evaluate the relationship between the packet reception ratio/BER and the link SNR *only* at the PHY layer, we would like to send a large number of frames and only evaluate the PRR/BER. I am not sure if the frame detection will introduce any "bias" to this purpose and that's why I am thinking about setting up a synced link (through a MIMO cable) to just evaluate the Tx and Rx modules without frame detection. Does this make sense to you or do you have any suggestions?

Also, if we want to send continuous streams, would it be easy and modular to replace the message strobe block with a continuous source (e.g., random source)?

Another question Q4: When testing the framework with USRPs, would other sampling rates other than 5/10/20 MSa/s work? I have been trying out different sampling rates (e.g., 1/2 MSa/s) but it seems like the length of the paddings on the Tx side has also to be adjusted to work for different sampling rates. Do you have any suggestions on how to pick the lengths of the paddings at the front and tail?

Thanks very much!

bastibl, 2019-02-18:
Q1: I don't see how this is supposed to work. Even with a MIMO cable, I don't think that you are automatically "synced". I would just measure with the sync blocks.

If you want to send a continuous stream, you could also pregenerate the samples (pipe them to a file sink) and stream the file. This way, you'd have exact interarrival times. Use the packet pad block to place some zeros between successive frames.

Q4: Yes, just set the same sample rate at sender and receiver. I mainly tested 5/10/20MHz and didn't have to adjust anything. What "padding" did you adjust? If, at all, this should only be required if you increase the sample rate over 20MHz (which I wouldn't recommend).

tingjunchen, 2019-02-19:
Q1: What did you mean by measure with the sync blocks? I was thinking about connecting 2 USRPs through a MIMO cable and use the function probe block to set the time and start time of the USRP sinks/sources. From my experiments, the delay between the two USRPs seems to be quite stable (10s of samples that can probably be calibrated out).

Also, I saw that you set the clock to be "synced to the PC clock" in the over-the-air examples, is there a particular reason for doing that or the "default" setting would also work?
 
Pregenerating the Tx samples does seem like a good approach and I will try it out.

Q4: I don't recall the exact numbers but my experiences were that for higher sampling rates, more paddings (pad_front or pad_tail or both) may need to be set to a larger number to make the frame detection work better...

Thanks,
Tingjun

bastibl, 2019-02-19:
Q1: 
> we don't care about the frame detection (at least at this point) and would still like to have the rest of the Rx functionalities (freq offset correction, demodulation/decoding), how would it work?

This sounds like you want to remove blocks from the flow graph. That's not so easy. I just recommended to use the flow graph as is. If SNR is so low that you couldn't detect the frame, you will not be able to decode subcarrier constellations anyway. So I would not worry about frame detection. If you worry about frequency offset correction, just hard code the value to 0 in the code and use the MIMO cable. That's much easier than trying to remove blocks.

I also don't get the plan for the function probe, but I still fail to understand what you are planning to do and how this is supposed to work.

The USRP was "synced to host" since the Packet Pad block allows delaying the frame for some time. To specify the time host and USRP have to use the same time. It's completely irrelevant for any DSP stuff and not used in the default flow graph.

Q4: Exactly. I was only irritated because you talked about lower samples rates and more padding. The amplifiers and oscillators take some time to settle, so the first samples are distorted. To delay the payload the same time, you have to add more padding for higher sample rates.

tingjunchen, 2019-02-21:
Thanks for the information - it's been really helpful. In fact, we are building full-duplex radios and are trying to use the gr-ieee802-11 framework for PHY/link (and potentially MAC) layer experimental evaluations.

One more question about the packet_len: I know how to compute the frame length with a given MCS and the paddings:

PHY_frame_len = ceil[ (16 + (pdu_len+28)*8 + 6)/(48 normalized to the MCS) ] * 80 + 160 (STF) + 160 (LTF) + 80 (SIG) + pad_front + pad_tail

But for some reason, the packet_len showing in a time sink GUI always seems to be exactly 1 sample longer than what I computed. Any suggestions?

bastibl, 2019-02-21:
The additional sample comes from the cyclic prefixer, which averages over two samples between OFDM symbols. It also adds one sample at the end of the frame.

tingjunchen, 2019-02-21:
Thanks a lot! I think I got most of my questions answered - I am closing this issue.

closed 2019-02-21 by tingjunchen