bastibl/gr-ieee802-11 #156: installed modules not found error while running "wifi_phy_hier"

khg2478, 2019-02-19:
Hi.

I have installed
- UHD and GNURadio using pybombs.
- Cloned "https://github.com/bastibl/gr-ieee802-11".
- Since I wanted to maintain dependency issues, I installed gr-foo by pybombs install gr-foo
- and I followed the installation steps; mkdir build, cmake .., make, sudo make install, sudo ldconfig.
and now I am facing this error when I run "wifi_phy_hier".

[Errors]
Block key "ieee802_11_chunks_to_symbols_xx" not found
Block key "ieee802_11_decode_mac" not found
Block key "ieee802_11_frame_equalizer" not found
Block key "ieee802_11_mapper" not found
Block key "ieee802_11_moving_average_xx" not found
Block key "ieee802_11_moving_average_xx" not found
Block key "ieee802_11_sync_long" not found
Block key "ieee802_11_sync_short" not found

Any suggestions?

It seems that all related files are installed in the location of gnuradio as in ~/prefix. I can see include, lib, and share directories containing built files of this library.

Thanks in advance.

bastibl, 2019-02-19:
The blocks ended up in a directory that GNU Radio doesn't know about. You have to adapt your `local_blocks_path`. See the installation instructions:
https://www.wime-project.net/installation/

khg2478, 2019-02-20:
@bastibl 

Thank you so much. I didn't know there is pybombs installation method to each of those modules. For people like me, "pybombs install gr-ieee-80211" will work.

Closing this silly issue.

closed 2019-02-20 by khg2478