bastibl/gr-ieee802-11 #157: swig conflict error

khg2478, 2019-02-20:
Hi again.

I just want to make sure I installed properly. So, I tested some of gr-ieee-80211 examples.

I was able to run wifi_phy_hier.grc and wifi_loopback.grc files using gnuradio_companion. However, other examples seem to conflict with swig module. Loading the the grc file is okay but running the program shows an error message as below.

openlte@openlte-Precision-T7600:~/prefix/default/src/gr-ieee-80211/examples$ gnuradio-companion wifi_transceiver.grc 
<<< Welcome to GNU Radio Companion 3.7.13.4 >>>

Block paths:
	/home/openlte/.grc_gnuradio
	/home/openlte/prefix/default/share/gnuradio/grc/blocks

Loading: "/home/openlte/prefix/default/src/gr-ieee-80211/examples/wifi_transceiver.grc"
>>> Done

Generating: '/home/openlte/prefix/default/src/gr-ieee-80211/examples/wifi_transceiver.py'

Executing: /usr/bin/python2 -u /home/openlte/prefix/default/src/gr-ieee-80211/examples/wifi_transceiver.py

MAPPER: encoding: 0
set_min_output_buffer on block 9 to 397056
set_min_output_buffer on block 11 to 397056
set_min_output_buffer on block 13 to 397056
set_min_output_buffer on block 14 to 397056
set_min_output_buffer on block 16 to 397056
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.15.0.git-13-g52138314
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Asking for clock rate 40.000000 MHz... 
[INFO] [B200] Actually got clock rate 40.000000 MHz.
Traceback (most recent call last):
  File "/home/openlte/prefix/default/src/gr-ieee-80211/examples/wifi_transceiver.py", line 361, in <module>
    main()
  File "/home/openlte/prefix/default/src/gr-ieee-80211/examples/wifi_transceiver.py", line 349, in main
    tb = top_block_cls()
  File "/home/openlte/prefix/default/src/gr-ieee-80211/examples/wifi_transceiver.py", line 185, in __init__
    self.uhd_usrp_source_0.set_auto_dc_offset("", 0)
  File "/home/openlte/prefix/default/lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 3502, in set_auto_dc_offset
    **return _uhd_swig.usrp_source_sptr_set_auto_dc_offset(self, enb, chan)
TypeError: in method 'usrp_source_sptr_set_auto_dc_offset', argument 2 of type 'bool'**

>>> Done

Any ideas?

Also, I wonder what is the safest way to install gnuradio. This time, I installed it with pybombs but "apt install gnuradio" seemed to be best because I didn't have this problem earlier.

Thank you so much.

bastibl, 2019-02-20:
Looks like UHD changed the interface for the SDR. I don't have one, so can't test it at the moment.

But click on UHD Source -> FE Corrections and replace `""` with `True`. At least that's what the error message suggests.

khg2478, 2019-02-20:
@bastibl 

Thanks for the reply.

I have revised the code as you recommended, replacing "" with True.

LN185: self.uhd_usrp_source_0.set_auto_dc_offset(True, 0)
LN186: self.uhd_usrp_source_0.set_auto_iq_balance(True, 0)

but it now complains that 

File "wifi_transceiver.py", line 25, in <module>
    from gnuradio import blocks
ImportError: No module named gnuradio

I can import gnuradio though. 

Python 2.7.12 (default, Nov 12 2018, 14:36:49) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import gnuradio
>>> 

It seems some dependency issue in pybombs.. Any ideas?

bastibl, 2019-02-20:
Probably an issue with your environment from which you started GRC. Try starting it from the shell where you started the Python shell.

khg2478, 2019-02-21:
@bastibl 
I am actually running the commands on the same shell. Here is the result of the execution. Could you please help? Because of this dependency issue, I installed the GNU Radio multiple times with different approaches. I am not sure why sometimes some part of grc code do not recognize gnuradio modules. Thank you so much as always.

openlte@openlte-Precision-T7600:~/prefix/default/src/gr-ieee-80211/examples$ python
Python 2.7.12 (default, Nov 12 2018, 14:36:49) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import gnuradio
>>> 
KeyboardInterrupt
>>> 
[6]+  Stopped                 python
openlte@openlte-Precision-T7600:~/prefix/default/src/gr-ieee-80211/examples$ sudo python wifi_transceiver.py 
Traceback (most recent call last):
  File "wifi_transceiver.py", line 25, in <module>
    from gnuradio import blocks
ImportError: No module named gnuradio
openlte@openlte-Precision-T7600:~/prefix/default/src/gr-ieee-80211/examples$

bastibl, 2019-02-21:
Do not use sudo. If you have permission problems, install udev rules. I guess

```
sudo python
import gnuradio
```

will also fail.

khg2478, 2019-02-21:
@bastibl 

Thank you for the reply again. Not using sudo still give me an error. I used it because of the error below.

openlte@openlte-Precision-T7600:~/prefix/default/src/gr-ieee-80211/examples$ python wifi_transceiver.py 
MAPPER: encoding: 0
set_min_output_buffer on block 9 to 397056
set_min_output_buffer on block 11 to 397056
set_min_output_buffer on block 13 to 397056
set_min_output_buffer on block 14 to 397056
set_min_output_buffer on block 16 to 397056
[INFO] [UHD] linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.15.0.git-13-g52138314
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Asking for clock rate 40.000000 MHz... 
[INFO] [B200] Actually got clock rate 40.000000 MHz.
set_min_output_buffer on block 32 to 100000
Traceback (most recent call last):
  File "wifi_transceiver.py", line 361, in <module>
    main()
  File "wifi_transceiver.py", line 349, in main
    tb = top_block_cls()
  File "wifi_transceiver.py", line 246, in __init__
    self.blocks_tuntap_pdu_0 = blocks.tuntap_pdu('tap1', 440, False)
  File "/home/openlte/prefix/default/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig6.py", line 1055, in make
    return _blocks_swig6.tuntap_pdu_make(dev, MTU, istunflag)
RuntimeError: gr::tuntap_pdu::make: tun_alloc failed **(are you running as root?)**

I installed udev rules according to the link "https://files.ettus.com/manual/page_transport.html".

cd <install-path>/lib/uhd/utils
sudo cp uhd-usrp.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules
sudo udevadm trigger

After all these, I still fail with the error above..

bastibl, 2019-02-21:
So you are using the tap interface. You'd have set it up beforehand as shown in `apps/rx_demo.sh`

```
sudo ip tuntap add dev tap0 user ${USER} mode tap
sudo ifconfig tap0 down
sudo ifconfig tap0 hw ether 12:34:56:78:90:ab
sudo ifconfig tap0 up
sudo ifconfig tap0 192.168.123.1
```

Then run as user. If that still doesn't work:
- check if
```
sudo python
import gnuradio
``` 
works
- disable tun/tap block in GRC and run as user

khg2478, 2019-03-03:
@bastibl 

It now works fine. :) Thanks a lot. I will close this issue.

closed 2019-03-03 by khg2478