bastibl/gr-ieee802-11 #158: Pad bits for data field

dbj4225, 2019-02-27:
Hi,

Seems no pad bits are added in generate_bits() function.
[image]
[image]
only service field is appended.

Can you please clarify if I am mistaken and some where else this phy padding is taken care?

bastibl, 2019-02-27:
Padding is already included in n_data_bits. See utils.cc/frame_param. This function mainly puts the payload at the right place.

dbj4225, 2019-02-27:
frame_param calculates the no of data bits including padding. But I could not see where appending of these padded bits actually  happens in utils.cc/frame_param.Is this done somewhere else?
[image]

Can you please help with this regard..

bastibl, 2019-02-27:
I don't understand the problem. How would you expect that this should look like? Like adding byte to the buffer? From this lines of code, you can see that `n_data_bits` includes padding. The code in the mapper allocates a buffer of size `n_data_bits` and copies the payload in the middle, i.e., not at the end. So there are some bits left at the end of the buffer, which is the padding.
Please take a bit more time to familiarize yourself with the code (and maybe C++). Your asking question with high frequency, all of them general WLAN or programming stuff. This is mainly an issue tracker for issues of the code.

closed 2019-03-01 by dbj4225