bastibl/gr-ieee802-11 #159: Network analysis and other protocols

KarlL2, 2019-03-04:
Hi,

I was wondering if this can be used as a network analysis tool, to list the different SSIDs, their frequencies, possibly their RSSIs. Eg if I can sample 100MHz of bandwidth, can I then process the whole bandwidth at once and detect all WiFi networks at once?

Also what about MIMO protocols like 802.11n? Can this module decode the SSID, or does it require additional code to handle it?

bastibl, 2019-03-04:
This is a physical layer implementation that can decode all IEEE 802.11a/g/p frames. To decode headers headers, you can export the recorded data as pcap and inspect the output with Tcpdump/Wireshark/Tshark and extract things like the SSID. (Actually, the SSID is one of the few things that is decoded within this modules. This is, however, mainly for debug. I'd recommend to use Wireshark).

Frequency is known, it's the one that you tuned the SDR to. RSSI is available in the trace too.

Regarding 100MHz: It's already pretty resource intensive to decode 20MHz in real-time. If you don't have a very powerful PC, I doubt that you will be able to decode more than 1 or 2 channels in parallel.

IEEE 802.11n (MIMO, LDPC, different preambles, etc.) are at the moment not implemented.

closed 2019-03-05 by KarlL2

Cobi-Coder, 2025-03-27:
I still don't understand, how do you measure the RSSI signal?

bastibl, 2025-03-27:
The Radiotap headers of the Wireshark file have fields for signal and noise. Since the SDR is not calibrated, there are no absolute values available. Only the SNR is estimated during decoding. It, therefore, sets these values to arbitrary values such that the SNR makes sense. See here:

https://github.com/bastibl/gr-foo/blob/maint-3.10/lib/wireshark_connector_impl.cc#L100-L117

Cobi-Coder, 2025-03-27:
What do you think if i calibrate with reference device.
Example:
True RSSI (Measured by Wi-Fi Adapter): -50 dBm
Measured RSSI from HackRF: 10 * log10(0.005) ≈ -23 dB
Calibration Factor = True RSSI - Measured HackRF RSSI= (-50) - (-23 = -27 dB
=> RSSI_calibrated = RSSI_HackRF + Calibration Factor

Cobi-Coder, 2025-03-28:
Thankyou