bastibl/gr-ieee802-11 #161: wifi_rx.grc limesdr can't find device

calgaryy, 2019-03-11:
Hi there, I'm trying to run this program on my LimeSDR to capture WiFi packets. However, when I run the program I get

`
RuntimeError:LookupError:KeyError:No devices found for ----->
Empty Device Address
`

Do I have to manually enter my device information or should it be detected automatically?

Using

Ubuntu 18.04
GRC-3.7.11
LimeSDR-Mini

bastibl, 2019-03-11:
The example flow graph use UHD sources/sinks, which work with SDRs from Ettus. You have to replace them with the corresponding source/sink blocks for your SDR.

calgaryy, 2019-03-11:
Thank you, I have the block here. 

The block asks for calibration bandwidths and analog filter bandwidths, do you know what these should be set at?

bastibl, 2019-03-11:
Sorry, but I don't have a Lime SDR. That's a question for Lime.

calgaryy, 2019-03-11:
Thanks very much. I'll close this issue 😊

closed 2019-03-11 by calgaryy