bastibl/gr-ieee802-11 #163: Regarding sampling rate and center frequency set in USRP

sayaz, 2019-03-13:
Hi,

A simple question (might be a stupid one as well !). But couldn't resist to ask. Please let me know where exactly I am missing the point.

In the transmit/receiver file, the **USRP sink/source** has a default value of 10MSPS sampling rate with 5.89GHz center frequency. I understand the term _sampling rate_ and _sampling frequency_ are same in GnuRadio. In order to recover the transmitted signal we must have the sampling frequency twice of the center frequency. That is for 5.89GHz I should have atleast 11.78GHz sampling frequency, 1178e9(sampling rate). I understand such sampling rate is absurd with USRPs.

I wanted to know where am I understanding it wrong ? Or the Gnuradio has different meaning on sampling rate, or does your flow has some mechanism for this?

sayaz, 2019-03-13:
Got the answer from someone. It can be closed.
So, the sampling should be twice the bandwidth. With complex sampling rate of 10MHz is good for 10MHz BW.

closed 2019-03-13 by sayaz

bastibl, 2019-03-14:
Exactly, and the other issue was center frequency vs bandwidth, i.e., a DSRC signal is sent at 5.9GHz and has a bandwidth of 10MHz. You don't have to sample the whole spectrum as the SDR mixes the signal down to baseband (by multiplying it with the center frequency, which is generated by the oscillator).