bastibl/gr-ieee802-11 #165: Receiving packet from router/AP

sayaz, 2019-03-18:
Hi Bastian,

Reference Issue #98
I have 2 TP-Link N750 routers with 3 antennas. I have configured them with channel 1 (2.412 GHz) 20MHz BW, and to operate on legacy mode. I can ping between them.

Now I am trying to use wifi_rx.grc to catch the packet while the ping is going on between the two routers. But I cannot receive any packets. When I open the Wireshark .pcap file, it says "The capture file appears to be damaged or corrupt". The only thing I have changed in USRP source is the Center frequency to be 2.412 GHz. (All the devices are kept within 3m range)

What I am trying to do is capture the packets (any kind of) that are being transmitted/received between the two routers.  Does the USRP require any kind of handshaking protocol with the routers to capture packets? Or any kind of association in required? What is the correct way of achieving this ?

bastibl, 2019-03-19:
The general setup is reasonable. I'm not sure what goes wrong. You could try opening a 11a (5GHz) AP since there is no backwards compatible mode on 5GHz. (what is also known to work is opening an ad hoc network with a MacBook). 

If it's not backwards compatibility mode, your SDR setup does not work. Check if there are underruns, change the gain, make sure the antenna is connected at the right port,  etc. etc. When testing over the air, start with 5GHz again.

(The USRP does not require any handshaking, it's just a passive sniffer)

sayaz, 2019-03-28:
Solved.
Router set to Legacy mode only with 2.5GHz & Increased the gain of the USRP

closed 2019-03-28 by sayaz