bastibl/gr-ieee802-11 #166: how to get CSI value?

townblack, 2019-03-27:
Dear Bastibl,
    Thanks for your gr-ieee802-11 project.I think I might be too new to this area. I feel very sorry to bother you.
I want to  get Channel State Information from wifi_rx,and I choose "LS" channel estimation algorithm in Wifi Frame Equalizer block.But there is nothing outputed.So I want to ask how can I get the CSI value.
    Thanks.

sayaz, 2019-03-28:
I had the same question to Bastian before, issue #130 

The CSI values are not logged. So you need to print them by editing the source code of "ls.cc" in /lib/equalizer/

Below is one example with "cout"
```
noise += std::pow(std::abs(d_H[i] - in[i]), 2);
signal += std::pow(std::abs(d_H[i] + in[i]), 2);
d_H[i] += in[i];
d_H[i] /= LONG[i] * gr_complex(2, 0);                               
cout << "CSI" << d_H[i] << endl;
}
d_snr = 10 * std::log10(signal / noise / 2);
```

townblack, 2019-03-28:
> I had the same question to Bastian before, issue #130
>
> The CSI values are not logged. So you need to print them by editing the source code of "ls.cc" in /lib/equalizer/
>
> Below is one example with [...]

townblack, 2019-03-28:
> I had the same question to Bastian before, issue #130
> The CSI values are not logged. So you need to print them by editing the source code of "ls.cc" in /lib/equalizer/
> Below is one example with [...]

Thank you, dear sayaz! I will have a try very soon.

closed 2019-03-28 by townblack

is353, 2019-08-18:
> I had the same question to Bastian before, issue #130
>
> The CSI values are not logged. So you need to print them by editing the source code of "ls.cc" in /lib/equalizer/
>
> Below is one example with [...]

i made the same changes to the file, but when i run the receiver, with log and debug enabled in the wifi frame equalizer, I am unable to see the print out in the terminal. Could you please let me know where you make the file. I am using gr-ieee802-11/build/ as the make location.

If I am missing a point please do feel free to fill in. 

Thank you

bastibl, 2019-08-19:
make sure you recompiled and installed the library after the change.