bastibl/gr-ieee802-11 #167: How Symbol alignment is done

sayaz, 2019-04-01:
Hi Bastian,

I need your help once more to understand how the symbol alignment is done.

As far as I understand, this is done with **Wifi Sync Long** block. After the frame start is detected with STF, the remaining of the samples are passed to this sync long block (including STF). 

The receiver knows the sample of LTF (in the time domain), this is (1/2 from last part(32) + full(64) + full(64)) = 160 samples. This is 2.5 time repetition of 64 samples.

Now, we are cross-correlating the received preambles (STF + LTF) with LTF (known) is this correct? Because when I am trying to plot the same in Matlab, I am getting a bit different result than your plot (in the paper). Can you please describe the procedure ?

bastibl, 2019-04-01:
What you say is correct. Hard to tell why your plot looks different. (You are doing *something* in Matlab that looks *a bit different*.) I'd recommend to check the SNR and the interface to the filter, it should cross-correlate with the complex conjugate of  the LTS in time domain. Some filter reverse/do not reverse the sequence. Some do a complex conjugate, some don't.

sayaz, 2019-04-05:
Bastian,

Need one suggestion for a different setup scenario.

The preambles are in Legacy mode in your setup. Say for example, if we want to change that in HT-mixed mode where the HT fields comes after the L-SIG, do we need to design the transmitter in similar fashion.

What I mean is, in your setup, 1st the L-SIG is BPSK modulated and then the L-STF/L-HTF are declared in OFDM allocator block. But in case of HT-Mixed mode, we have other fields after L-SIG, in this case do we need to declare them before we declare the L-SIG ? Or is theire any way that we can delcare all the L-STF/L-LTF/HT-STF/HT-LTF in the same OFDM allocator block. 

Though the question is not direclty related to your flow, but I wanted to know what would you have done for such scenario?
[image]

bastibl, 2019-04-06:
In the current implementation, I tried to use the GNU Radio upstream OFDM blocks when possible. AFAIS, this more complex scenario is not supported, so you'd have to implement custom blocks for this frame format.

sayaz, 2019-04-09:
Thanks for the previous comment.

Another question: I understand the equation for calculating CSI. Where you are using both the LTF OFDM symbols and dividing by the known symbols at Rx. 

Say a router is transmitting a packet that has a payload size of x(random). In this case how are you extracting the position of the LTFs to calculate the CSI? For example, we are receiving a frame after wifi_sync_long block : LTF(64) + LTF (64) + SIG (64) + Payload(x) and this is repeating. Now to calculate CSI, we need both the LTFs. Now fort the first frame I know the position of the 1st LTF is from 0 to 64, but what will be the position of the 2nd frame to extract the LTF? Are you using the tagged 'wifi_frame_start' in such case ? Can you please briefly explain this?

bastibl, 2019-04-09:
Yes, the `wifi_frame_start` tag indicates that the symbol is the first symbol of a frame, i.e., the first LTF. Downstream blocks recognize the tag and reset/resychronize, i.e., calculate the CSI and decode the signal field.

sayaz, 2019-04-19:
Hi Bastian, 

2 Questions!

Question on tag: "wifi_start". This is first used in sync._short.cc. I just wanted to be sure, if the tag is set exactly at the start of the frame that is at the first samples of STF or it is set when after the threshold is detected for continuous MIN_Plateau=2. 

Below is the time domain plot after Wifi_sync block. I did edit the preambles (adding 2 new preambles, 1 OFDM symbol each), total 8 OFDM symbols. The STF and LTF are exactly the same that you provided. As I have increased the length of the preamble, I have also changed the MIN_GAP = 640 & MAX_SAMPLES = 640*80 accordingly. The complete frame is detected and passed wifi sync short block from where I am getting this o/p. I am confused as I am seeing the tag after the first few samples of STF and not at the very start. Therefore I wanted to know is it because of the change I made, or it is correct? I apologize for not understanding the code accurately.
[image]

My second question, the Wifi_Sync_long block after symbol alignment removes the CP, outputs from start of LTF (removes STF). This time the position of the tag is also changed that is at the start of LTF. 

If I plot this, the signal seems very short (though we don't have CP and STF). How can I be definite that it is actually passing every sample from the start of LTF till Data symbols(including my additional preambles in the middle)? Also as I am sure that the sync short block is passing my whole frame, that means the value I changed for MAX_SAMPLES and MAX_GAP are working correctly, is there any other place that I should look at?
[image]

bastibl, 2019-04-21:
> Question on tag: "wifi_start". This is first used in sync._short.cc. I just wanted to be sure, if the tag is set exactly at the start of the frame that is at the first samples of STF or it is set when after the threshold is detected for continuous MIN_Plateau=2.

This is not possible. At this stage the frame is just detected and not precisely aligned. This is what SyncLong is for. The wifi_start tag is set by SyncShort whenever the autocorrelation crosses the threshold (for two successive samples). This can happen anytime during the plateau.

> My second question, the Wifi_Sync_long block after symbol alignment removes the CP, outputs from start of LTF (removes STF). This time the position of the tag is also changed that is at the start of LTF.

Yes, because the short training sequence is not forwarded to downstream blocks.

> If I plot this, the signal seems very short (though we don't have CP and STF). How can I be definite that it is actually passing every sample from the start of LTF till Data symbols(including my [...]

Looks like you send short frames back to back, i.e., there is another preamble before MAX_SAMPLES are processed, which triggers resynchronization. Since you know what you are sending, you need to do the math and calculate how many samples you would expect and check.

> If I extract the LTF after FFT (forward block), I am getting values like (8.875xxxxx+1.13xxxxj) instead of (1+0j) due to the scaling done at FFT_Reverse block (tuple[1/52**0.5]*64), I wanted to know while calculating the CSI, how is this taken care of?

The receiver has to be able to process signals independent from absolute power levels. It calculates an initial estimate of the channel using the LTF and uses it to scale data symbols. See the frame equalizer block.

closed 2019-06-06 by bastibl