bastibl/gr-ieee802-11 #170: wifi_rx.grc not parsing frames in the 2.4GHz range

zgodwin1, 2019-04-29:
Hello, ive been trying to use wifi_rx to parse frames in the the 2.4ghz range. Ive tried a number of things but it doesnt seem like its possible with the wifi_rx flow graph. I can pick up beacons in the 5ghz range no problem. Is this because 2.4ghz band is too crowded?

 Also despite running the volk profile, checking to make sure my usb driver and cable/port are usb 3.0, running the flow graph with an intel i7 w/32gb ram, and checking for DC offset im still receiving overflow and no frames in the 2.4ghz range. please help!!!
[image]

zgodwin1, 2019-04-29:
Follow up: I also noticed i can parse frames at 5ghz with a 20MHz sample rate but not at 22, 25, or 40MHz. Shouldnt it be possible to collect packets with any sample rate greater than 20MHz?

bastibl, 2019-04-30:
You cannot configure arbitrary sample rates, as the receiver expects a given number of samples per symbol.  You could increase the sample rate and downsample in software but there's no real benefit.
The frames on 2.4GHz are probably not real 11g frames but 11b or 11g in backwards compatible mode. If you want to test reception in 2.4GHz you could, for example, open an Ad Hoc network with a MacBook (they send real 11g frames) or configure an AP to send beacons in 11g-only mode.

zgodwin1, 2019-04-30:
Thank you, much appreciated.

closed 2019-05-01 by zgodwin1