bastibl/gr-ieee802-11 #172: "ImportError: no module name _ieee802_11_swig" when running as root

francisgalloway, 2019-06-08:
Curious error I've encountered. When I run a gnuradio-companion generated python script as root, it gives me `ImportError: no module named _ieee802_11_swig`

My paths seems to be ok; LD_LIBRARY_PATH is `/usr/local/lib64`, PYTHONPATH is `/usr/local/lib64/python2.7/site-packages`. The _ieee802_11_swig.so file is properly in `/usr/local/lib64/python2.7/site-packages/ieee802_11/`.
I added a `print sys.path` to the beginning of the script to verify and, sure enough, the PYTHONPATH is in there. But I'm getting the ImportError nonetheless. Any thoughts?

Specifically, I'm trying to run the wifi_transceiver example, which I have to run as root because it includes tuntap. Running as a non-root user gives me a tuntap error telling me to run as root. 

Any thoughts?

bastibl, 2019-06-11:
Running it as root is bad practice, but I guess you know :-)

Anyhow, make sure that `libgnuradio-ieee802-11.so` is under `/usr/local/lib64` and that you ran `sudo ldconfig` after installing the library.

If the problem persists, it usually indicates a problem with libraries. Unfortunately, Swig hides these errors. Edit the installed `ieee802_11_swig.py`. At the top you should see something like:

```
try:
    return importlib.import_module(mname)
except ImportError:
    return importlib.import_module('_ieee802_11_swig')
```

Change it to
``` 
except ImportError as er:
    print(er)
     return importlib.import_module('_ieee802_11_swig')
```

francisgalloway, 2019-06-11:
Solved. I copied the .so file into /usr/lib64/python2.7, and after that it was able to find it.

Still have errors, but they need another  this one is solved.

francisgalloway, 2019-06-11:
^sorry, misclicked reopen. I'm gonna open another issue for the other error.

closed 2019-06-11 by francisgalloway