bastibl/gr-ieee802-11 #173: wifi_transceiver permissions: does this need to be run as root?

francisgalloway, 2019-06-11:
So, I'm working with the wifi_transceiver example, trying to modify it for the HackRF SDR. I've consistently hit a certain error when trying to run the flowgraph as non-root:

`Traceback (most recent call last):
  File "./wifi_transceiver_hackrf.py", line 343, in <module>
    main()
  File "./wifi_transceiver_hackrf.py", line 331, in main
    tb = top_block_cls()
  File "./wifi_transceiver_hackrf.py", line 236, in __init__
    self.blocks_tuntap_pdu_0 = blocks.tuntap_pdu('tap0', 440, False)
  File "/usr/lib64/python2.7/site-packages/gnuradio/blocks/blocks_swig6.py", line 1046, in make
    return _blocks_swig6.tuntap_pdu_make(dev, MTU, istunflag)
RuntimeError: gr::tuntap_pdu::make: tun_alloc failed (are you running as root?)`

If I bypass the TUNTAP_PDU block, it runs just fine. And some of the discussion surrounding this block indicates that it requires root privileges to run. 

However, when I DO run the executable as root, I get different error:
`X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x109`
I don't think this is an issue with the HackRF, since I'm able to run an fm receiver as root without any issues. So I'm thinking that this flowgraph wasn't meant to be run as root in the first place. Is that the case? Is there a way around the tun_alloc error that doesn't involve root privileges?

francisgalloway, 2019-06-11:
Update: I tried the wifi_rx.grc, and that seems to be working fine. Swapped out USRP source for osmocom, and it all runs fine. Only gives me malformed packets in the wifi.pcap, though.

bastibl, 2019-06-12:
Can you please summarize again what the current problem is? Why do you think the packets are malformed and how did you inspect them?

francisgalloway, 2019-06-13:
When I open the output file, wifi.pcap, in Wireshark, it only appears to contain 20 "malformed" packets. All received at t=0, with 0 bytes on the wire/captured. 
[image]

The hexdump shows the same few bytes repeated 20 times: `c3d4 a1b2 0002 0004 0000 0000 0000 0000 ffff 0000 007f 0000`

The same thing seems to happen when I disconnect the radio, or swap my osmocom source for a null source. So I've clearly screwed something up on the input side. I think I may have translated the gain wrong; originally, I assumed that the "normalized gain" parameter on the USRP input meant non-db gain, so I took 20log10(normalized gain) and plugged that into the osmocom's "RF gain" parameter. Further research has revealed that "normalized gain" means something different, so that gain calculation was flawed. I also tried using the default values for RF/IF/BB gain for the osmocom source, but that predictably failed. 

It's also possible that my frequency/bandwidth is off, since I'm not really confident in my calculations. Regardless, these are more "HackRF" issues than "gr-ieee802_11" ones. My initial question was about the permissions for wifi_transceiver.py, and I've realized that wifi_rx is more appropriate, since HackRF is half-duplex.

bastibl, 2019-06-14:
To test if the code works as it is supposed to, use the loopback flowgraph to generate a wifi.pcap file and check its content. I've never seen such frames in the trace. Please make sure that you are using the unmodified source. Also i you use a HackRF, you will have to compensate the DC offset in the center of the spectrum (otherwise it will trigger frame detection constantly and overload the PC).

francisgalloway, 2019-06-14:
Solved! Switched the default sample rate to 20MHz, and the default frequency to one of the 2.4GHz bands. Not sure why changing the sample rate was necessary, but it works.

For the record, my wifi_loopback had been running perfectly fine, and producing the correct output.

bastibl, 2019-06-17:
Glad it worked. You have to switch to 20MHz because 11g WLAN in der 2.4GHz band uses 20MHz channels.

closed 2019-06-17 by bastibl