bastibl/gr-ieee802-11 #174: Fail to execute the wifi_tx

lochana-chathura, 2019-06-28:
Hello,

We tried to run the wifi_tx of your model. The wifi_loopback is working fine. But when the wifi_tx is run, the following error code is generated.

 __See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=288000
[33;1m[WARNING] [UHD] [39;0mUnable to set the thread priority. Performance may be negatively affected.
Please see the general application notes in the manual for instructions.
**EnvironmentError: OSError: error in pthread_setschedparam__**

We already ran sudo sysctl -w net.core.wmem_max=288000
But still get the same error.

Please help us on this matter.

Thank You.

bastibl, 2019-07-01:
This warnings and the error are not from the WLAN module but from UHD. Your issue seems to be covered in the readme of the driver: https://files.ettus.com/manual/page_general.html

lochana-chathura, 2019-07-01:
sorry. the error got disappeared after running gnuradio-companion with sudo. Thanks for the response :)

closed 2019-07-01 by lochana-chathura

bastibl, 2019-07-01:
Great it worked. But I wouldn't recommend to run this as root. Just allows your user to configure the thread priority, by changing `/etc/security/limits.conf` as explained in the wiki.

lochana-chathura, 2019-07-01:
noted. will work on it. thank you.