bastibl/gr-ieee802-11 #175: Can't build maint-v.37 branch on ARMv8

akshayy29, 2019-07-03:
I'm having the same exact issue posted here: https://github.com/bastibl/gr-ieee802-11/issues/120#issue-334109703 .

```
I'm trying to build this on a Raspberry Pi, but make fails with the following:
[ 7%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/moving_average_cc_impl.cc.o c++: error: unrecognized command line option '-msse2'

Presumably, the issue is that ARM doesn't have -msse2. I commented out the msse2 line from CMakeLists.txt and I get a different (also x86-specific) error:
In file included from /root/sdr/gr-ieee802-11/lib/decode_mac.cc:20:0: /root/sdr/gr-ieee802-11/lib/viterbi_decoder.h:20:10: fatal error: xmmintrin.h: No such file or directory #include <xmmintrin.h> 
```

I also tried contacting rcetin, since he solved the issue using cross compilation & modifying the sources, but haven't got any reply from him yet.

bastibl, 2019-07-09:
Not sure what exactly you commented out, but can you please try commenting out this line:

akshayy29, 2019-07-11:
You're referring us to maint-3.8, but we are using maint-3.7,

bastibl, 2019-07-15:
maint-3.7 doesn't have this compile switch. Either "back port" the generic implementation to 3.7 (copy-pasting a file and some cmake), switch to 3.8, or use an older version of the 3.7 branch that worked with GNU Radio next.

dattajigosavi, 2019-07-23:
We followed the steps provided in Jetson Nano device.The generic implementation was copied in the CMakeLists.txt in the "lib" folder .
[image]
[image]

After executing the make command on the terminal says the 100% completion. But when we tried to use the examples we got a error states  "block  not found" and ieee 802_11 not defined.
[image]
[image]

What is your advice to solve this error?

bastibl, 2019-07-24:
The hierarchical block has to be build. After that blocks have to be reloaded or GRC has to be restarted. Please see the installation instructions:
https://www.wime-project.net/installation/

tranzmatt, 2019-12-17:
It makes no sense to hardcode Intel's "sse2" flag with GNURadio being built on ARM devices.

closed 2019-12-21 by bastibl

gmartina, 2020-04-08:
How can you build the hierachical block in an embedded platform with just gnuradio, without gnuradio companion? 

thankyou

bastibl, 2020-04-08:
you can build it with `grcc` which is installed as part of GNU Radio.

gmartina, 2020-04-08:
i dont have gnuradio companion installed in my embedded device. Do you mean i can build it in a desktop PC and use it in my embedded device?

gmartina, 2020-04-08:
(i meant i dont have grcc installed, i have only gnuradio instaled, minimal image, no graphics)

bastibl, 2020-04-08:
`grcc` does not use graphics. Anyway, if it is not available, you'd have to build the block on the host and copy it over. The  generated Python script and Block definition are usually under `~/.grc_gnuradio`).

gmartina, 2020-04-08:
@bastibl  Thank you for you fast reply!