bastibl/gr-ieee802-11 #176: not right message from wifi_loopback.grc

girs1982, 2019-07-05:
i see that :
[image]

duration: 00 00
frame control: 00 08 (DATA)
Subtype: Data
seq nr: 76
mac 1: 42:42:42:42:42:42
mac 2: 23:23:23:23:23:23
mac 3: ff:ff:ff:ff:ff:ff
instantaneous fer: 0
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

new mac frame  (length 795)
=========================================
duration: 00 00
frame control: 00 08 (DATA)
Subtype: Data
seq nr: 77
mac 1: 42:42:42:42:42:42
mac 2: 23:23:23:23:23:23
mac 3: ff:ff:ff:ff:ff:ff
instantaneous fer: 0
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

new mac frame  (length 795)
=========================================
duration: 00 00
frame control: 00 08 (DATA)
Subtype: Data
seq nr: 78
mac 1: 42:42:42:42:42:42
mac 2: 23:23:23:23:23:23
mac 3: ff:ff:ff:ff:ff:ff
instantaneous fer: 0
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
MAPPER: encoding: 2

new mac frame  (length 795)
=========================================
duration: 00 00
frame control: 00 08 (DATA)
Subtype: Data
seq nr: 79
mac 1: 42:42:42:42:42:42
mac 2: 23:23:23:23:23:23
mac 3: ff:ff:ff:ff:ff:ff
instantaneous fer: 0
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

new mac frame  (length 795)
=========================================
duration: 00 00
frame control: 00 08 (DATA)
Subtype: Data
seq nr: 80
mac 1: 42:42:42:42:42:42
mac 2: 23:23:23:23:23:23
mac 3: ff:ff:ff:ff:ff:ff
instantaneous fer: 0
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
MAPPER: encoding: 0

new mac frame  (length 795)
=========================================
duration: 00 00
frame control: 00 08 (DATA)
Subtype: Data
seq nr: 81
mac 1: 42:42:42:42:42:42
mac 2: 23:23:23:23:23:23
mac 3: ff:ff:ff:ff:ff:ff
instantaneous fer: 0
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

new mac frame  (length 795)
=========================================
duration: 00 00
frame control: 00 08 (DATA)
Subtype: Data
seq nr: 82
mac 1: 42:42:42:42:42:42
mac 2: 23:23:23:23:23:23
mac 3: ff:ff:ff:ff:ff:ff
instantaneous fer: 0
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

new mac frame  (length 795)
=========================================
duration: 00 00
frame control: 00 08 (DATA)
Subtype: Data
seq nr: 83
mac 1: 42:42:42:42:42:42
mac 2: 23:23:23:23:23:23
mac 3: ff:ff:ff:ff:ff:ff
instantaneous fer: 0
xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
how it done?

bastibl, 2019-07-09:
I'm not sure what you expected, but that looks good to me. The demo flow graph just sends a bunch of 'X's. You can change it to something different in the Message Strobe block.

girs1982, 2019-07-09:
in sollution wrote was hello world,and how i can fecode wifi?

bastibl, 2019-07-09:
Use the flow graphs that interface hardware. To receive WiFi, start with the wifi_rx flow graph.

girs1982, 2019-07-09:
i try it,but saw only XXXXXXXXXXX,UUUUUU,and nothing,was try change frequnce ,but no has results(((

bastibl, 2019-07-09:
Can you please describe in much more detail what you actually did and what happened.

girs1982, 2019-07-09:
yes, small later,i try decode my router, and some wifi signal,but not understand and not saw how it work...

girs1982, 2019-07-09:
try run wifi rx ,have errors:
Traceback (most recent call last):
  File "/home/ubludok/gr-ieee802-11/examples/wifi_rx.py", line 373, in <module>
    main()
  File "/home/ubludok/gr-ieee802-11/examples/wifi_rx.py", line 361, in main
    tb = top_block_cls()
  File "/home/ubludok/gr-ieee802-11/examples/wifi_rx.py", line 150, in __init__
    self.uhd_usrp_source_0.set_auto_dc_offset("", 0)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 3457, in set_auto_dc_offset
    return _uhd_swig.usrp_source_sptr_set_auto_dc_offset(self, enb, chan)
TypeError: in method 'usrp_source_sptr_set_auto_dc_offset', argument 2 of type 'bool'

bastibl, 2019-07-09:
That's a problem of the UHD block. Double-click the USRP Source block and change DC Offset under the FE Corrections tab. Looks like it wants a bool value. So put `True` or `False` there.

girs1982, 2019-07-09:
done it,but:
UHD Warning:
    Unable to set the thread priority. Performance may be negatively affected.
    Please see the general application notes in the manual for instructions.
    EnvironmentError: OSError: error in pthread_setschedparam
Traceback (most recent call last):
  File "/home/ubludok/gr-ieee802-11/examples/wifi_rx.py", line 374, in <module>
    main()
  File "/home/ubludok/gr-ieee802-11/examples/wifi_rx.py", line 362, in main
    tb = top_block_cls()
  File "/home/ubludok/gr-ieee802-11/examples/wifi_rx.py", line 151, in __init__
    self.uhd_usrp_source_0.set_auto_dc_offset("true", 0)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 3457, in set_auto_dc_offset
    return _uhd_swig.usrp_source_sptr_set_auto_dc_offset(self, enb, chan)
TypeError: in method 'usrp_source_sptr_set_auto_dc_offset', argument 2 of type 'bool'

bastibl, 2019-07-09:
Set it to `True` without the quotes...

girs1982, 2019-07-09:
UHD Warning:
    Unable to set the thread priority. Performance may be negatively affected.
    Please see the general application notes in the manual for instructions.
    EnvironmentError: OSError: error in pthread_setschedparam
Traceback (most recent call last):
  File "/home/ubludok/gr-ieee802-11/examples/wifi_rx.py", line 374, in <module>
    main()
  File "/home/ubludok/gr-ieee802-11/examples/wifi_rx.py", line 362, in main
    tb = top_block_cls()
  File "/home/ubludok/gr-ieee802-11/examples/wifi_rx.py", line 152, in __init__
    self.uhd_usrp_source_0.set_auto_iq_balance("", 0)
  File "/usr/local/lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 3475, in set_auto_iq_balance
    return _uhd_swig.usrp_source_sptr_set_auto_iq_balance(self, enb, chan)
TypeError: in method 'usrp_source_sptr_set_auto_iq_balance', argument 2 of type 'bool'

bastibl, 2019-07-09:
...and now you do the same thing for the other argument in the tab.

girs1982, 2019-07-09:
I launch )))
[image]
but how decode???

bastibl, 2019-07-09:
Set sample rate to 20Mhz, tune to a channel where 11a or 11g frames are sent. Try different gain settings to see what performs best.

girs1982, 2019-07-09:
i get it
frame control: 00 94 (CONTROL)
Subtype: Block ACK
RA: d4:6e:e:2:c6:60
TA: e4:18:6b:55:d6:78
DLONG: frame start at 320
DLONG: frame start at 320
LONG: frame start at 165
LONG: frame start at 170
encoding: 0 - length: 37 - symbols: 14
encoding: 0 - length: 14 - symbols: 6
length: 33

new mac frame  (length 33)
=========================================
duration: 00 3c
frame control: 00 d0 (MANAGEMENT)
Subtype: Action
seq nr: 2169
mac 1: e4:18:6b:55:d6:78
mac 2: cc:50:e3:5d:a8:e3
mac 3: e4:18:6b:55:d6:78
length: 10

new mac frame  (length 10)
=========================================
frame too short to parse (<20)
LONG: frame start at 169
LONG: frame start at 164
encoding: 0 - length: 67 - symbols: 24
length: 63

new mac frame  (length 63)
=========================================
duration: 00 00
frame control: 00 40 (MANAGEMENT)
Subtype: Probe Request
seq nr: 2103
mac 1: ff:ff:ff:ff:ff:ff
mac 2: 7c:dd:90:e2:d3:0
mac 3: ff:ff:ff:ff:ff:ff
encoding: 0 - length: 14 - symbols: 6
length: 10

new mac frame  (length 10)
=========================================
frame too short to parse (<20)
LONG: frame start at 171
LONG: frame start at 171
encoding: 4 - length: 14 - symbols: 2
length: 10

new mac frame  (length 10)
=========================================
frame too short to parse (<20)
LONG: frame start at 320
LONG: frame start at 170
encoding: 0 - length: 32 - symbols: 12
length: 28

new mac frame  (length 28)
=========================================
duration: 00 00
frame control: 00 94 (CONTROL)
Subtype: Block ACK
RA: e4:18:6b:55:d6:78
TA: d4:6e:e:2:c6:60
LONG: frame start at 170
encoding: 4 - length: 20 - symbols: 2
length: 16
[... 548 lines cut ...]
frame too short to parse (<20)
LONG: frame start at 163
LONG: frame start at 170
encoding: 0 - length: 92 - symbols: 32
length: 88

new mac frame  (length 88)
=========================================
duration: 00 3c
frame control: 49 08 (DATA)
Subtype: Data
seq nr: 3194
mac 1: e4:18:6b:55:d6:78
mac 2: cc:50:e3:5d:a8:e3
mac 3: e4:18:6b:55:d6:78
instantaneous fer: 0.5
yL. '....g...Z.]...Q.w.O.!.h..tyO.Al\$M..FJP.~.k/#4.._/T.{....M.
encoding: 0 - length: 14 - symbols: 6
length: 10

new mac frame  (length 10)
=========================================
frame too short to parse (<20)
D
>>> Done

girs1982, 2019-07-09:
and how up AP,and name wifi AP?

girs1982, 2019-07-09:
i can done it 
[image]

bastibl, 2019-07-09:
Great, happy to hear that it works.

closed 2019-07-09 by girs1982