bastibl/gr-ieee802-11 #177: how send deauth paket to mac adress

girs1982, 2019-07-15:
how send deauth paket to mac adress??? i think it posible

girs1982, 2019-07-15:
et_min_output_buffer on block 32 to 100000
Traceback (most recent call last):
  File "/home/ubludok/gr-ieee802-11/examples/wifi_transceiver.py", line 354, in <module>
    main()
  File "/home/ubludok/gr-ieee802-11/examples/wifi_transceiver.py", line 342, in main
    tb = top_block_cls()
  File "/home/ubludok/gr-ieee802-11/examples/wifi_transceiver.py", line 240, in __init__
    self.blocks_tuntap_pdu_0 = blocks.tuntap_pdu("tap0", 440, False)
  File "/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig5.py", line 2059, in make
    return _blocks_swig5.tuntap_pdu_make(dev, MTU, istunflag)
RuntimeError: gr::tuntap_pdu::make: tun_alloc failed (are you running as root?)

bastibl, 2019-07-16:
You don't need the TUN/TAP interface to send a deauth frame. Use Scapy. This slides show hot to send  a Beacon frame, but there is  a similar function for sending a death frame.
https://fosdem.org/2019/schedule/event/gr_scapy/attachments/slides/3366/export/events/attachments/gr_scapy/slides/3366/gnuradio_meets_scapy.pdf

If you want to use the TUN/TAP interface nevertheless, checkout the shell scripts in the apps folder.

girs1982, 2019-07-16:
wifi_transceiver: 

et_min_output_buffer для блока 32 до 100000 
Traceback (последний вызов был последним): 
файл "/home/ubludok/gr-ieee802-11/examples/wifi_transceiver.py", строка 354, в файле 
main () main /) 
"/ home / ubludok / gr- ieee802-11 / examples / wifi_transceiver.py ", строка 342, в основном 
файле tb = top_block_cls () 
" /home/ubludok/gr-ieee802-11/examples/wifi_transceiver.py ", строка 240, в файле init
self.blocks_tuntap_pdu_0 = Файл blocks.tuntap_pdu ("tap0", 440, False) 
"/usr/lib/python2.7/dist-packages/gnuradio/blocks/blocks_swig5.py", строка 2059, в make 
return _blocks_swig5.tuntap_pdu_make (dev, MTU, istunflag)
Ошибка RuntimeError: gr :: tuntap_pdu :: make: tun_alloc (вы работаете от имени root?)
how done it ???

bastibl, 2019-07-16:
see the shell scripts in the apps folder and try to understand how this is supposed to work. Or just start it as root, but that will just make the error go away. You'll still have zero functionality without understanding the networks setup. So please familiarize yourself with the shell scripts in the apps folder.

girs1982, 2019-07-16:
ok,but if you help me,will cool

girs1982, 2019-07-16:
in scrapy have errors     /Dot11Beacon()/Dot11Elt(ID='SSID', info='GR WLAN')
    ^
IndentationError: unexpected indent

bastibl, 2019-07-16:
The indentation error is from Python. Please use Google.
The link works for me.

girs1982, 2019-07-16:
i start it,and wait result,it many spaces betwen strigs

bastibl, 2019-07-16:
I have no idea what you expect from me. Can you please describe in detail and with actual sentences what you are doing and what is not working as expected.

girs1982, 2019-07-16:
i need to create simple wifi deauther,i find code https://www.bastibl.net/gnuradio-wlan-scapy/  ,copy paste and it have errors,next i delete spaces and edit text,it was launch, i see in gnu radio 
tGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtGtG

closed 2019-07-16 by bastibl