bastibl/gr-ieee802-11 #182: I cant see any packets wlan0 monitor mode!!!

Gonzaless-Chura, 2019-07-24:
i run tx example,but nothing not see in monitor wlan!!!
[image]

bastibl, 2019-07-25:
Please see the `rx_demo.sh` script in the apps folder. Also you started Wireshark to listen on you normal WLAN card. You can either log to file and open the file in wireshark (as shown in the example flow graph), create a FIFO and read from the FIFO as shown in the `rx_demo.sh` script or create a TUN/TAP interface and listen on this interface (also shown in the `rx_demo.sh` script.

Gonzaless-Chura, 2019-07-25:
what need to do ,for see packets standard wlan monitor???, i create usrp_wifi_txr,like esp8266 jammer,but not fully understand yours job!!!

bastibl, 2019-07-25:
Again, please checkout and use the `rx_demo.sh` script. If you have any specific questions, feel free to ask. But please stop this "hlp plx, thing not working!!!"-questions.

Gonzaless-Chura, 2019-07-25:
i need send death packet in my wifi AP,how i can it done,use yours job???
nothing not packets
[image]

bastibl, 2019-07-25:
...............

closed 2019-07-25 by bastibl