bastibl/gr-ieee802-11 #183: Selecting values for LO Offset

sayaz, 2019-07-26:
Hi Bastian,

I have changed few things in the flow graph at the receiver after FFT. I have included my own phase correction block instead of Wifi Equalizer. After decoding, i have successfully recovered a sending file with 16-QAM in a simulation.  From this I am able to say that everything is working perfectly.

Now, I replace the sink/source with USRP (B210) one used as Rx and the other as Tx, kept 2m apart. Here I have failed to receive the file (empty/ sometimes garbage). 

Troubleshoot that I have performed:
- I have checked that I have clean transmit signal by using a spectrum analyzer (2.412 GHz). 20 MSPS.
- The constellation points (16-QAM) at receiver after FFT & phase correction is jumping and not stable at all.
- The packets are detected correctly at the wifi_sync_short block. Also tested with padding.

From my point of view there is LO_offset present. I have tried with different values from 0 to 20MHz. But nothing changes. My question is how can i accurately find the value for this offset ? And do you have any suggestion where should I look into?

bastibl, 2019-07-26:
From my experience, the fact that something work in simulations, doesn't imply that it works perfectly also with real world transmission with hardware impairments.

That being said, there will be no perfect, stable constellation points immediately after the FFT. There is residual frequency offset and channel effects that have to be compensated in the equalizer. The constellation points displayed in the example flow graphs after after equalization. Make sure that your algorithm deals with these effects. Otherwise it will only work im simulations.

sayaz, 2019-07-26:
Yes of-course, after FFT, I have calculated the CSI by extracting one of the sub-carrier (k=10th) from LTF1 and LTF2, added them, and divide by (2* known LTF of the same sub-carrier) (Least Square method). 

For phase offset correction, I have used this single CSI value and inverse over one full frame. Did this for every incoming frames (LS Equalizer). After doing so, the constellation points were stable in simulation. I am yet to try other algorithms.

The reason I was thinking about LO offset is because in my simulation, I have used a channel model block with noise voltage = 0.2. and 0.0 for frequency offset. At this setting everything works fine. But when I change the Freq offset to 0.1 or higher, the constellation point starts jumping around, similar way with USRPs. So I was hoping to know how to choose the LO offset value? Do i do it by seeing any plot ? Can you provide one example?

bastibl, 2019-07-29:
Sorry, but I still don't get it. You extract CSI from one subcarrier out of 52 and use it to correct the whole channel? This won't work in practice.

I also don't get why you expect frequency offset correction to be broken. I assume it works without your changes. (Use git to make sure that you use the unmodified code.)

Also what's the difference between "oscillator offset" and "frequency offset"? WLAN uses a single oscillator for the sample clock and to synthesize the center frequency. The oscillator offset causes the frequency offset.

And yes, for some (more extreme) values of the channel and offsets, things will break. I don't have the exact numbers, but why don't you compare the initial implementation with your estimator and check the differences.

> -So I was hoping to know how to choose the LO offset value?

Which LO offset value? For the simulations?

> -In your loopback flow, how did you came to the equation of freq off in channel model block.

The channel model block doesn't know about the sample rate. So if you want to have a given frequency offset, you have to calculate the corresponding phase rotation between successive samples.

sayaz, 2019-07-29:
1 frame for example (After sync_long block) = LTF(64) + LTF(64) + SIG(64) + Data(128). I am calculating CSI from one sub-carrier and using it to correct this whole frame(SIG(64)+Data(128))). And again repeating this process for a new frame. Are you saying this is not the right procedure ? If not, do I need to correct symbols by computing CSI for every sub-carrier separately?

Yes, the simulation runs fine. The problem was some specific values on channel model block. 

The problem arises when I run with USRP. What LO offset value should I choose in USRP ? And how do I find that out (both for Tx and Rx)?

bastibl, 2019-07-29:
> Are you saying this is not the right procedure ?

Yes.

>  If not, do I need to correct symbols by computing CSI for every sub-carrier separately?

Yes. (Some equalizers also do some advanced stuff like averaging in frequency domain.)

> What LO offset value should I choose in USRP ?

I think you are confusing several things. The LO offset option of UHD is for offset tuning to tune the LO out of the band of interest, i.e., the LO is not the center frequency. The offset should, therefore, depend on the bandwidth of the signal. See the UHD manual.

sayaz, 2019-08-14:
Dear Bastian,

In OFDM carrier allocator block, the LTF1 is the rotated version that is created in sync_word.py. At receiver, while calculating CSI = LTF1 + LTF2/(LTF-known), how the rotation is compensated here? What I mean is at receiver LTF1 is different from LTF2, so how do we calculate CSI in this respect? And what will be the known LTF at the receiver, LTF1 or LTF2 ? Can you please explain?

bastibl, 2019-08-14:
Please see this issue https://github.com/bastibl/gr-ieee802-11/issues/154

sayaz, 2019-08-19:
Hello,
Apologies, I am having some confusion.

In carrier allocator, we have LTF1 + LTF2. 
LTF1 here is generated from sync_words.py which is long_rot.
LTF2 is as per standard.
After passing through CP it is = 1/2LTF + LTF1 + LTF2 = 160 samples.

Now at receiver, wifi_sync_long block get rids of the CP, which is 1/2 LTF in this case. The remaining, LTF1+LTF2 is passed through.

For estimating CSI with LS method we need LTF1 to be the same as LTF2. As LTF1 here is actually rotated, we need to reverse this rotation.

Does this reverse of rotation is happening in the wifi_sync_long block? if so can you point me to the line?

bastibl, 2019-08-22:
The signal in the air is not rotated (otherwise it wouldn't be standard compliant). The reasons the samples in the carrier allocator are rotated is because they are afterwards piped through the cyclic prefix block. You can see the addition of a cyclic prefix as a rotation, which is why the samples are first rotated in the other direction. That's a bit ugly, but I wanted to use the upstream block :-)

sayaz, 2019-08-22:
Thank you Bastian, I now understand. I confused the rotation of the symbol that was not happening over the air. So over the air we have as per the standard. Therefore at receiver, we estimate CSI with the LTF1+LTF2/2* known LTF. Where LTF1 and LTF2 are the same just after the CP block at TX.

closed 2019-12-21 by bastibl