bastibl/gr-ieee802-11 #184: Start tag expected, '<' not found

r00tb3, 2019-08-04:
I'm constantly getting the following error when I open or un the `.grc` files from the example folder.

`gr-ieee802-15-4/examples/transceiver_OQPSK.grc:1:1:FATAL:PARSER:ERR_DOCUMENT_EMPTY: Start tag expected, '<' not found`

What does this mean?

Pls, help.

bastibl, 2019-08-05:
Looks like you are using the 3.8 version of the flow graph with a 3.7 installation of GNU Radio. If you use GNU Radio 3.7, please use the main-3.7 branch of this module, as suggested in the readme.

bastibl, 2019-08-05:
Discussion seems to continue in the correct repository
https://github.com/bastibl/gr-ieee802-15-4/issues/44

closed 2019-08-05 by bastibl

r00tb3, 2019-08-05:
> Looks like you are using the 3.8 version of the flow graph with a 3.7 installation of GNU Radio. If you use GNU Radio 3.7, please use the main-3.7 branch of this module, as suggested in the readme.

I've installed this module using pybombs.

If I install manually I get compilation errors and it doesn't work.

I've tried building this module and the 802.15.4 using the maint-3.7 and maint-3.8 and master branch but it fails and none of it works.