bastibl/gr-ieee802-11 #185: sampling clock offset

WangQiwei, 2019-08-07:
Hi, @bastibl 
I know that pilot tracking is used to correct for imperfections in the equalizer response, and there are three kinds of pilot tracking: Phase tracking, Amplitude tracking, and Timing tracking.
It seems only phase tracking is used in this receiver example. If I want to add the time tracking function, which part of the code should I modify? 

Thanks

bastibl, 2019-08-07:
Can you please explain a little more detailed what you are planning to do? With time tracking, you mean the sample clock offset? Do you really want to track it (based on the comb pilots or feedback from data subcarriers?) or only estimate it at the beginning of the frame? The latter is already done.

WangQiwei, 2019-08-12:
Yes, I hope to use the pilot tracking function of the pilot tone to track and correct the sampling timing deviation more finely.

bastibl, 2019-08-12:
The time alignment before the FFT is done by the Sync Long block to decide over which window to do the FFT. This is symbol alignment.
You can annotate information with GNU Radio's stream tags. This is already used heavily, for example to communicate frequency offset to downstream blocks. You can either check the code of the GNU Radio documentation.

WangQiwei, 2019-08-12:
Your suggestion helps me a lot. Thank you very much.

closed 2019-12-21 by bastibl