bastibl/gr-ieee802-11 #186: no rx answer on my usrp

VIRUSOLDLIu, 2019-08-08:
constellation
i use my own usrp device to test the rx/tx module ,but sometimes it works sometimes it not (i did nothing between the two phenomenon). when i can not receive the message, it still shows a clear constellation.
and the terminal always answers "long :frame start at 320".
i wonder how it occurs .

bastibl, 2019-08-12:
This message indicates that symbol alignment might have gone wrong, i.e., it couldn't find the correct position of the symbols from the long training sequence. This can have many reasons. The RF environment might have changed (interference), the channel changed (maybe adjust gain) etc.

VIRUSOLDLIu, 2019-08-15:
Well，I think there may got something wrong between the the usrp devices and rx codes . Now there is another question that when I work the tx program on N210, it flows U all the time . While the tx works well on other devices.I found U means "PC not providing data quickly". I wonder how it happens . My cpu works only 70% overall, which i think  is still capable of sending wifi signals.

bastibl, 2019-08-15:
There shouldn't be any CPU problems. The flow graph uses a tagged streams, i.e., all samples of a frame are computed before they are streamed to the device. This sounds like problems with the networking setup.
Connect the SDR directly (no switch), don't use a VM, apply the network-related settings recommended in the Ettus Wiki:
http://files.ettus.com/manual/page_usrp2.html

closed 2019-12-21 by bastibl