bastibl/gr-ieee802-11 #191: Unable to CMake in Ubuntu 18.04

harrihaven2, 2019-09-18:
Hey there,

I recently updated my GNURadio to 3.8 as per [these instructions](https://wiki.gnuradio.org/index.php/UbuntuInstall) due to an error I would get from attempting to cmake this project according to the directions in the readme. But even after the upgrade I am unable to make the project and open the .grc example files. When I try the GNU-Companion looks for an xml file and gets upset when it doesn't see angle brackets in the wifi_*.grc files.

This is the cmake error:

```
-- PyBOMBS installed GNU Radio. Setting CMAKE_INSTALL_PREFIX to /home/ltmcmill/gnuradio
-- Build type not specified: defaulting to release.
CMake Error at CMakeLists.txt:84 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.8".

  The following configuration files were considered but not accepted:

    /home/<user>/gnuradio/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.7.13.5
    /home/<user>/sdr/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.7.13.5

-- Configuring incomplete, errors occurred!
```

Is my GNURadio update not modifying some files it needs to, or what am I doing wrong?

Thank you for your help

bastibl, 2019-09-18:
It looks like you have two version of GNU Radio installed. Please try uninstalling 3.7.

harrihaven2, 2019-09-19:
Ah, thank you! As a follow up question, I installed the latest version of GNURadio (3.9), but 3.8 is required by this repo. Can I change the requirement to accept 3.9 or will that break everything?

bastibl, 2019-09-19:
You cannot arbitrary mix branches. As indicated in the readme, you'll have to use maint-3.7 with maint-3.7 of GNU Radio, or maint-3.8 with maint-3.8 of GNU Radio, or master with master of GNU Radio. Any other combination will not work.

harrihaven2, 2019-09-19:
Understood. Thank you for your help!

closed 2019-09-19 by harrihaven2